Fabian-Schmidt/esphome-truma_inetbox #2: Truma & Victron on one ESP?

snipah, 2023-02-26:
Hi Fabian,

Just switched from inetbox2mqtt to your implementation because I prefer the hassle free config/update-procedure via Wireguard from everywhere! Everything is working as expected! Thanks for your great work!

I understood, that I might be able to run your Truma and Victon-Implmentation on one ESP! I currently have https://github.com/KinDR007/VictronMPPT-ESPHOME running on a separate 8266 for a Shunt and Smart Solar. Can I run the victrun-stuff to the same ESP your Truma-implementation is running on?

I expanded my Truma.yaml-file with the Victron-config:

`esphome:
  name: "esphome-truma"

ota:

api:

external_components:
  - source: github://Fabian-Schmidt/esphome-truma_inetbox
  - source: github://Fabian-Schmidt/esphome-victron_ble

logger:
  level: DEBUG

esp32:
  board: esp32dev
  framework:
    type: arduino

esp32_ble_tracker:

victron_ble:
  - id: MySmartShunt
    mac_address: "E7:B0:D9:6d:62:03"    
    bindkey: "e289af4fb6c8ed333494297acc0bb637"
  - id: MySmartSolar
    mac_address: "58:10:C2:BD:7D:05"
    bindkey: "3af563e56794c0fb9198f541af2f08a9"

uart:
  - id: lin_uart_bus
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600
    data_bits: 8
    parity: NONE
    stop_bits: 2

truma_inetbox:
  uart_id: lin_uart_bus
  
binary_sensor:
  - platform: truma_inetbox
    name: "CP Plus alive"
    type: CP_PLUS_CONNECTED
  - platform: truma_inetbox
    name: "Room Heater active"
    type: HEATER_ROOM
    id: HEATER_ROOM
  - platform: truma_inetbox
    name: "Water Heater active"
    type: HEATER_WATER
    id: HEATER_WATER
  - platform: truma_inetbox
    name: "Timer active"
    type: TIMER_ACTIVE
    id: TIMER_ACTIVE
  - platform: truma_inetbox
    name: "Timer Room Heater active"
    type: TIMER_ROOM
  - platform: truma_inetbox
    name: "Timer Water Heater active"
[... 576 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for 192.168.1.29
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for 192.168.1.29: Error connecting to ('192.168.1.29', 6053): [Errno 111] Connect call failed ('192.168.1.29', 6053)
INFO Trying to reconnect to 192.168.1.29 in the background
`

Fabian-Schmidt, 2023-02-27:
Thanks for the logs.
I am looking into the issue.
It appears that logging is a blocking operation. So when a log is send via Wifi it takes too long. And the Lin listener is using an interrupt which than takes too long.

I developed and tested only via USB connection which has no issue.

I should have an update in a day or two.

snipah, 2023-02-27:
Thanks! BLE received no data during my tests. Just by looking at the logs: Is everything connecting correctly regarding the Victron devices? I have seen nothing obvious in the logs...

Fabian-Schmidt, 2023-03-01:
Hi,
I have uploaded a new version which should fix this issue.
Can you please test again?

snipah, 2023-03-01:
I have the feeling, that once I activate logging, the connection breaks. If I deactivate logging, everything works without problems. But I just tested the Truma part.

Fabian-Schmidt, 2023-03-01:
Try setting refresh on the component to ensure you have the latest version e.g.

```yaml
  - source: github://Fabian-Schmidt/esphome-truma_inetbox
    components: ["truma_inetbox", "uart"]
    refresh: always
```

closed 2023-03-02 by snipah