Fabian-Schmidt/esphome-truma_inetbox #5: Home Assistant config

snipah, 2023-03-02:
Hey Fabian!

I guess that you are using Home Assistant for your implementation if the Truma Inetbox.

Could you possibly share your config for Home Assistant as some kind of blue print? I'm asking, because not everybody is as talented in creating SW and GUI-elements.

Fabian-Schmidt, 2023-03-02:
Hi,
I am currently using my ESP Home standalone without Home Assistant. I don't even have a HA instance running anywhere.
I might look into it to provide guidance, but I have no idea about HA or it's GUI elements. But this won't happen soon.

Fabian-Schmidt, 2023-03-02:
mc0110 has created a nice UI for HA which can maybe be taken as a basis
https://github.com/mc0110/inetbox2mqtt/tree/main/doc

snipah, 2023-03-02:
Thanks! Will work something out!

closed 2023-03-02 by snipah

7xy, 2025-01-30:
Maybe the description of my solution in my blog (german) may help implementing it in Home Assistant! [https://schleeh.de/2025/01/24/schritt-fuer-schritt-truma-heizung-mit-esphome-einrichten/]