Fabian-Schmidt/esphome-truma_inetbox #6: Control ventilator (Off/Eco/Comfort/Boost) without heating (Temp = Zero)

snipah, 2023-03-02:
Hi Fabian,

is it possible to control the ventilator without setting hvac_modes to "heat"? Ideally you could control the ventilator independently of the hvac.

Fabian-Schmidt, 2023-03-02:
Hi,
I have not found an option to activate fan only.

maseb24, 2023-03-02:
Hello Fabian,
this is a great integration in HA, ESPHome.

I have already gained experience with inetbox 2 mqtt, so I was curious about your implementation.

I installed everything on my esp32. I was just playing around with it and I had a few questions.
With the previous inetboxes, you had to set the heating mode to off, eco or high for the Truma to start heating up. Where can I do that with you? The fan cannot be set separately.
My heating mode display is -1-.
You got the timers. Where can I put these

That's what I noticed first.
I really like your implementation.
Unfortunately, the website is not available. Did I do something wrong there?

Greetings Rudi

dignifiedquire, 2023-03-07:
It is possible on the panel manually, so I would expect there to be some option. It also seems there are some hints on setting the fan level on the decoding here: https://docs.google.com/spreadsheets/d/16nw7vJGL2m406Nb3y3_9kxW-dGib02PLaZ2sJC5xSz4/edit

Fabian-Schmidt, 2023-03-07:
I have not found any information in the communication between the Inetbox (emulation) and CP Plus. The CP Plus is the master in the communication.
Yes there must be communication between CP Plus and Combi 4 for activating the fan, but I am not interfering this communication. I can observe it, but not alter it.

To control the heating / CP Plus I have implemented three options:

 - Climate component
 - Number component
 - various Actions

For the  Timer I have added an action in ESPHome. I am unsure how to expose this using standard components to HA (Please let me know if you have idea / seen in it in other components). The issue is that I require 3 information to submit a valid message (start & stop time and room temperature).
Example using Action:

```yaml 
      - truma_inetbox.timer.activate:
          start: 7:00
          stop: 9:30
          # Required: Set room temp to a number between 5 and 30
          room_temperature: 13
          # Optional: Set heating mode: `"OFF"`, `ECO`, `HIGH`, `BOOST`
          heating_mode: ECO
          # Optional: Set water temp as number: `0`, `40`, `60`, `80`
          water_temperature: 0
          # Optional: Set energy mix to: `GAS`, `MIX`, `ELECTRICITY`
          energy_mix: GAS
          # Optional: Set electricity level to `0`, `900`, `1800`
          watt: 0
```

snipah, 2023-03-26:
I would consider this solved! As you can effectively set the temperature to 5 degrees and control the fan separately. With the recent changes even in HomeKit the fan becomes a separate entity, which is great.

closed 2023-03-26 by snipah

snipah, 2024-11-22:
> @snipah Could you please share how you do it? Thanks!

Do what? I think everything is explained above...