Fabian-Schmidt/esphome-truma_inetbox #7: CP Plus alive continuously switches from "On" to "Off"

snipah, 2023-03-02:
Just saw, that CP Plus alive status switches from "On" to "Off" all the time... Log shows nothing obvious. Can you think of any reason?

esp_poessl CP Plus alive  eingeschaltet
18:33:30 - Vor 13 Sekunden

esp_poessl CP Plus alive  ausgeschaltet
18:33:14 - Vor 29 Sekunden

esp_poessl CP Plus alive  eingeschaltet
18:32:29 - Vor 1 Minute

esp_poessl CP Plus alive  ausgeschaltet
18:32:12 - Vor 2 Minuten

esp_poessl CP Plus alive  eingeschaltet
18:31:27 - Vor 2 Minuten

esp_poessl CP Plus alive  ausgeschaltet
18:31:10 - Vor 3 Minuten

esp_poessl CP Plus alive  eingeschaltet
18:30:25 - Vor 3 Minuten

esp_poessl CP Plus alive  ausgeschaltet
18:30:08 - Vor 4 Minuten

esp_poessl CP Plus alive  eingeschaltet
18:29:23 - Vor 4 Minuten

closed 2023-03-02 by Fabian-Schmidt

Fabian-Schmidt, 2023-03-02:
I only allowed CP Plus 30 seconds of inactivity. But when no heating is active it will go to sleep for up to 60 seconds.
I increased the time to 90 seconds.

snipah, 2023-03-02:
Wow! That was a quick fix! Thanks

commit ccb7b956 (2023-03-02): Sleep time of CP Plus is ~60seconds allow 90 seconds before reporting an issue. Fix #7