Fabian-Schmidt/esphome-truma_inetbox #8: Function 'truma_inetbox.timer.disable' does not disable TIMER_ROOM

snipah, 2023-03-07:
As the title says:  'truma_inetbox.timer.disable' does not disable TIMER_ROOM

Fabian-Schmidt, 2023-03-07:
Can you please attach a log with at least `DEBUG` log level?
I have just tested the function with my Combi 4 and it is working.

snipah, 2023-03-08:
Hi Fabian,

See log below! Just the be precise: The function does not disable to binary_sensor *TIMER ROOM HEATER ACTIVE".

At least for me, it remains on after switching the timer from ON to OFF.

```
INFO Reading configuration /config/esphome/esp-poessl.yaml...
INFO Updating https://github.com/Fabian-Schmidt/esphome-truma_inetbox.git@None
INFO Updating https://github.com/Fabian-Schmidt/esphome-victron_ble.git@None
INFO Starting log output from 192.168.1.29 using esphome API
INFO Successfully connected to 192.168.1.29
[09:20:20][I][app:102]: ESPHome version 2023.3.0-dev compiled on Mar  6 2023, 23:17:13
[09:20:20][C][wifi:504]: WiFi:
[09:20:20][C][wifi:362]:   Local MAC: 0C:B8:15:CD:8F:F4
[09:20:20][C][wifi:363]:   SSID: 'poessl.net'[redacted]
[09:20:20][C][wifi:364]:   IP Address: 192.168.1.29
[09:20:20][C][wifi:366]:   BSSID: 94:18:65:0A:A6:C5[redacted]
[09:20:20][C][wifi:367]:   Hostname: 'esp-poessl'
[09:20:21][C][wifi:369]:   Signal strength: -51 dB ▂▄▆█
[09:20:21][C][wifi:373]:   Channel: 13
[09:20:21][C][wifi:374]:   Subnet: 255.255.255.0
[09:20:21][C][wifi:375]:   Gateway: 192.168.1.1
[09:20:21][C][wifi:376]:   DNS1: 0.0.0.0
[09:20:21][C][logger:296]:   Hardware UART: UART0
[09:20:21][C][uart.arduino_esp32:108]: UART Bus 1:
[09:20:21][C][uart.arduino_esp32:109]:   TX Pin: GPIO17
[09:20:21][C][uart.arduino_esp32:110]:   RX Pin: GPIO16
[09:20:21][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[09:20:21][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[09:20:21][C][uart.arduino_esp32:115]:   Data Bits: 8
[09:20:21][C][uart.arduino_esp32:116]:   Parity: NONE
[09:20:21][C][uart.arduino_esp32:117]:   Stop bits: 2
[09:20:21][C][template.sensor:023]: Template Sensor 'shunt_WATT'
[09:20:21][C][template.sensor:023]:   State Class: ''
[09:20:21][C][template.sensor:023]:   Unit of Measurement: ''
[09:20:21][C][template.sensor:023]:   Accuracy Decimals: 1
[09:20:21][C][template.sensor:024]:   Update Interval: 60.0s
[09:20:21][C][template.sensor:023]: Template Sensor 'charge'
[09:20:21][C][template.sensor:023]:   Device Class: 'energy'
[09:20:21][C][template.sensor:023]:   State Class: 'total_increasing'
[09:20:21][C][template.sensor:023]:   Unit of Measurement: 'kWh'
[09:20:21][C][template.sensor:023]:   Accuracy Decimals: 3
[09:20:21][C][template.sensor:024]:   Update Interval: 60.0s
[09:20:21][C][template.sensor:023]: Template Sensor 'discharge'
[09:20:21][C][template.sensor:023]:   Device Class: 'energy'
[09:20:21][C][template.sensor:023]:   State Class: 'total_increasing'
[09:20:21][C][template.sensor:023]:   Unit of Measurement: 'kWh'
[09:20:21][C][template.sensor:023]:   Accuracy Decimals: 3
[09:20:21][C][template.sensor:024]:   Update Interval: 60.0s
[09:20:21][C][template.switch:076]: Template Switch 'Activate Room Heater'
[09:20:21][C][template.switch:099]:   Restore Mode: restore defaults to OFF
[09:20:21][C][template.switch:060]:   Restore State: NO
[09:20:21][C][template.switch:061]:   Optimistic: NO
[09:20:21][C][template.switch:076]: Template Switch 'Activate Water Heater'
[09:20:21][C][template.switch:099]:   Restore Mode: restore defaults to OFF
[09:20:21][C][template.switch:060]:   Restore State: NO
[09:20:21][C][template.switch:061]:   Optimistic: NO
[09:20:21][C][template.switch:076]: Template Switch 'Activate Water Heater (enum)'
[09:20:21][C][template.switch:099]:   Restore Mode: restore defaults to OFF
[09:20:21][C][template.switch:060]:   Restore State: NO
[09:20:21][C][template.switch:061]:   Optimistic: NO
[09:20:21][C][template.switch:076]: Template Switch 'Active Timer'
[09:20:21][C][template.switch:099]:   Restore Mode: restore defaults to OFF
[09:20:21][C][template.switch:060]:   Restore State: NO
[09:20:21][C][template.switch:061]:   Optimistic: NO
[09:20:21][C][victron_ble:014]: Victorn BLE:
[09:20:21][C][victron_ble:015]:   Address: FD:55:52:B0:DF:2D
[09:20:21][C][victron_ble:015]:   Address: D0:4D:6B:30:48:1E
[09:20:21][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[... 864 lines cut ...]
[09:27:20][D][truma_inetbox.TrumaiNetBoxApp:413]: StatusFrameClock 09:27:00
[09:27:20][D][truma_inetbox.LinBusProtocol:227]: Multi package response FB
[09:27:20][D][victron_ble:247]: [FD:55:52:B0:DF:2D] Recieved BATTERY_MONITOR message.
[09:27:21][D][victron_ble:247]: [FD:55:52:B0:DF:2D] Recieved BATTERY_MONITOR message.
[09:27:21][D][victron_ble:242]: [D0:4D:6B:30:48:1E] Recieved SOLAR_CHARGER message.
[09:27:21][D][victron_ble:247]: [FD:55:52:B0:DF:2D] Recieved BATTERY_MONITOR message.
[09:27:22][D][victron_ble:242]: [D0:4D:6B:30:48:1E] Recieved SOLAR_CHARGER message.
[09:27:23][D][sensor:127]: 'discharge': Sending state 0.00000 kWh with 3 decimals of accuracy
[09:27:24][D][sensor:127]: 'Battery Voltage': Sending state 14.72000 V with 2 decimals of accuracy
[09:27:24][D][sensor:127]: 'Battery Current': Sending state 0.00000 A with 3 decimals of accuracy
[09:27:24][D][sensor:127]: 'Yield Today': Sending state 0.00000 kWh with 2 decimals of accuracy
[09:27:24][D][sensor:127]: 'PV Power': Sending state 0.00000 W with 0 decimals of accuracy
[09:27:24][D][sensor:127]: 'Load Current': Sending state -0.00000 A with 1 decimals of accuracy
[09:27:24][D][sensor:127]: 'Battery Power SS': Sending state 0.00000 W with 2 decimals of accuracy
[09:27:24][D][text_sensor:067]: 'MPPT state': Sending state 'Off'
[09:27:24][D][text_sensor:067]: 'MPPT Error reason': Sending state ''
[09:27:25][D][sensor:127]: 'Time remaining': Sending state 65535.00000 min with 0 decimals of accuracy
[... cut ...]
[09:27:25][D][sensor:127]: 'Starter Battery': Sending state 13.67000 V with 2 decimals of accuracy
[09:27:25][D][sensor:127]: 'Current': Sending state 1.55300 A with 3 decimals of accuracy
[09:27:25][D][sensor:127]: 'State of charge': Sending state 100.00000 % with 1 decimals of accuracy
[09:27:25][D][sensor:127]: 'Battery Power': Sending state 22.82910 W with 2 decimals of accuracy
[09:27:25][D][sensor:127]: 'Consumed Ah': Sending state -0.00000 Ah with 1 decimals of accuracy
[09:27:25][D][text_sensor:067]: 'Battery Alarm reason': Sending state ''
```

Fabian-Schmidt, 2023-03-08:
The component does not assumes the new state of the timer. It waits for the feedback from the CP Plus that the update was accepted.
This occurs in your log:

```txt
[09:21:27][D][binary_sensor:036]: 'Timer active': Sending state OFF
[09:21:27][D][switch:056]: 'Active Timer': Sending state OFF
```

If you perform the changes via the web interface: I have observed issues with the state of the web site not updating for some types of components and a refresh of the web site is necessary.
Alternative you could also set the flag `Optimistic` of the switch so it assumes that the change was made.

closed 2023-03-10 by snipah

danistst, 2025-01-03:
@snipah since I can't find another way to get in touch with you, I comment here.

According to your logs you also seem to have the victron ble integration and other things running on the same ESP.

What kind of ESP do you have? No performance problems?

I'm getting problems with truma_inetbox when I add two ble integrations and a level gauge via mcu6050.

snipah, 2025-01-04:
@danistst I am using an ESP32 Wroom. No performance problems here. What problems are you experiencing?

danistst, 2025-01-04:
Browsing through the open and closes issues here I realize that it might not be performance problems. But issues like the ESP connection dropping and reconnecting, or having to wait quite a few minutes before Truma control actually works. I will give it another try with all features and integrations on the same ESP and wait much longer before testing. I also have a ESP32 Wroom 32D.

danistst, 2025-01-05:
@snipah I've configured all features on a single ESP now and it works. The issue was the logging. I always had the page of the ESPHome open to see the logs. But this was creating issues. So once I closed that tab, it worked well.

May I ask you what kind of Wifi router you're using that the Home Assistant device and the ESP connect to? I'm using a ESP NAT router, which seems to be great in terms of energy consumption. But I have some stability issues of the network.

btw, is there a place where we could have such kind of discussions rather than misusing this issue here?