Fabian-Schmidt/esphome-truma_inetbox #12: generic LIN

swifty99, 2023-03-19:
Hi Fabian,

this is not really an issue, more a proposal.
I am really impressed of the code and the encapsulation. It looks like a lot of effort has been put in the driver.

Since a  long time I wanted to write a LIN driver especially for ESPhome. With this excellent code I started to build something like the CAN bus component for the LIN.

I can write assemblercode a lot easier than latest C++. So I would like to ask if you could help out with a code review for a generic component? 
A generic LIN would be a dream as tons off cheap hardwre from scrapped cars would be available for esphome like sensors, shunts, motors, pumps, fans ....

you can find my repo here:
https://github.com/swifty99/linbus

not in the best state yet, steep learning python and c++ at the same time...

Fabian-Schmidt, 2023-03-19:
I had a generic driver in mind when writing this, but I don't have access to any Lin devices.
The current state has some Truma specifics baked into the low level. As the Truma Tin Bus is not 100% compliant with the Lin Bus specs.
Also this is (and will stay) a Lin Client piece. I am aware of the challenges a Lin Master implementation must solve.

swifty99, 2023-03-19:
yes, the truma app on top of a generix LIN in Esphome would be no fun, and seems to work anyway.
I would be happy, if you could look over the generic part every now and then. I know to write assembly, C++ ist still very fancy.
I have a lot of LIN devices left over from scrapping cars.. I think you live in Berlin? No problem if you like to get some.
And these parts are way easier to control, set the right byte and of they go.