Fabian-Schmidt/esphome-truma_inetbox #16: Problems adding MQTT Broker

ubarni, 2023-07-04:
Hi,
i successfully connected my ESP to my Truma and was able to control it from the Web UI.
Now i wanted to add anything to my MQTT Server.
I use this in several ESPHOME projects and it worked out of the box:

```
mqtt:
  broker: !secret mqtt_url -> "192.168.178.11"
  port: !secret mqtt_port -> "1883"
  discovery: false
  topic_prefix: service/truma/set
```

But what i got in the Web UI Console is:

```
19:02:28 [D] [mqtt:183]   Resolving MQTT broker IP address...
19:01:14	[D]	[mqtt:216]	Resolved broker IP address to 185.198.140.245
19:01:14	[I]	[mqtt:246]	Connecting to MQTT...
19:01:32	[W]	[mqtt:335]	MQTT Disconnected: TCP disconnected.
19:01:32	[I]	[mqtt:246]	Connecting to MQTT...
19:01:51	[W]	[mqtt:335]	MQTT Disconnected: TCP disconnected.
19:01:51	[I]	[mqtt:246]	Connecting to MQTT...
19:02:09	[W]	[mqtt:335]	MQTT Disconnected: TCP disconnected.
19:02:09	[I]	[mqtt:246]	Connecting to MQTT...
19:02:28	[W]	[mqtt:335]	MQTT Disconnected: TCP disconnected.
```

Is there a way to configure the MQTT Broker?

Regards...
Uwe

ubarni, 2023-07-04:
What i want to achieve is to control my Truma in my Camper Van from my Home Assistant at home.
So i have to send the data from my local Camper Van WiFi to my MQTT Server at home.

Fabian-Schmidt, 2023-08-18:
Hi,
I finally found some time and installed an MQTT broker on my OpenWRT router to test. So far it is working for me without any issues.
Can you ensure there is no network related issue?

dougle03, 2024-02-17:
> What i want to achieve is to control my Truma in my Camper Van from my Home Assistant at home.
> So i have to send the data from my local Camper Van WiFi to my MQTT Server at home.

Did you get this sorted? The MQTT element in ESPHome is independent of the Truma components. From your logs the ESP is unable to find your MQTT server. Ensure you are able to reach your MQTT from the local wifi network your ESP is connected to. Failing that, try posting your MQQT issue on the main ESPHome page or the FB group