Fabian-Schmidt/esphome-truma_inetbox #19: Clearing error codes

gedger, 2023-10-15:
I have this connected to my Truma Combi 2e, appears to work fine.

Is there anyway of clearing an error code like you can from the display? On the display you use the rotating knob to move to the error code, press it and the code clears so the boiler will have another go. When I switch to gas and the isolating valve has been switched off (or I've forgot to switch the gas bottle on) it takes a couple of attempts before the gas gets through and then everything works fine. 

thanks

Fabian-Schmidt, 2023-10-15:
I am not aware of such feature.
This would be something to play around with the code and protocol.
As the protocol is not officially documented and instead only reverse engineered it is possible that somewhere is a hidden function to achieve this behavior.

gedger, 2023-10-16:
Thanks, I'll investigate further. May not be solvable as the display is the master device so may not be possible from a slave. I'll close and report back if I find anything.

closed 2023-10-16 by gedger