Fabian-Schmidt/esphome-truma_inetbox #20: Experience Combi 4E, CRC-Errors

gmanic, 2023-11-01:
First of all - thanks for this wonderful ESP implementation to get the Truma "under control".
My setup is a Combi 4E (C4.01.02 / 50.00, H1.01.00 / 340.0A) together with a dedicated esp32. Further sensors are planned.

I got the connection, although the Stop_bits 2 was a bit irritating, all other solutions have 1 stop-bit, as far as I remember.

Currently, the esp is not yet integrated in my HA system, as that is in a different subnet, but planned with a pi4 in the RV.

I do get some CRC errors, see below in my mqtt-messages. But as far as I can tell, all messages received reflect the status of the Truma.

```
truman-linbus/debug [D][sensor:094]: 'electric power level': Sending state 0.00000 W with 0 decimals of accuracy
truman-linbus/sensor/electric_power_level/state 0
truman-linbus/debug [D][sensor:094]: 'Energy mix': Sending state 1.00000  with 0 decimals of accuracy
truman-linbus/sensor/energy_mix/state 1
truman-linbus/debug [D][sensor:094]: 'Operating status': Sending state 6.00000  with 0 decimals of accuracy
truman-linbus/sensor/operating_status/state 6
truman-linbus/debug [D][sensor:094]: 'Heater error code': Sending state 0.00000  with 0 decimals of accuracy
truman-linbus/sensor/heater_error_code/state 0
truman-linbus/debug [W][component:214]: Component truma_inetbox took a long time for an operation (0.19 s).
truman-linbus/debug [W][component:215]: Components should block for at most 20-30ms.
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.18.3D.00.53.00.00.00.00.00.00.00.00.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00 (41)
truman-linbus/debug [I][truma_inetbox.TrumaiNetBoxApp:254]: StatusFrameTimer
truman-linbus/debug [D][truma_inetbox.TrumaiNetBoxAppTimer:064]: StatusFrameTimer target_temp_room: nan target_temp_water: nan 00:00 -> 00:00  OFF
truman-linbus/debug [D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [D][binary_sensor:036]: 'Timer active': Sending state OFF
truman-linbus/binary_sensor/timer_active/state OFF
truman-linbus/debug [D][binary_sensor:036]: 'Timer Room Heater active': Sending state OFF
truman-linbus/binary_sensor/timer_room_heater_active/state OFF
truman-linbus/debug [D][binary_sensor:036]: 'Timer Water Heater active': Sending state OFF
truman-linbus/binary_sensor/timer_water_heater_active/state OFF
truman-linbus/debug [E][truma_inetbox.LinBusListener:356]: PID 3D      order - unable to send response
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [D][truma_inetbox.LinBusProtocol:226]: Multi package request BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.17.00.0F.06.01.B4.0A.AA.0A.00.00.00. 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
truman-linbus/debug [I][truma_inetbox.TrumaiNetBoxApp:272]: StatusFrameConfig
truman-linbus/debug [D][truma_inetbox.TrumaiNetBoxAppConfig:011]: StatusFrameConfig Offset: 0.0
truman-linbus/debug [D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
truman-linbus/debug [D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.15.00.50.0D.2B.00.01.01.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
truman-linbus/debug [I][truma_inetbox.TrumaiNetBoxApp:263]: StatusFrameClock
truman-linbus/debug [D][truma_inetbox.TrumaiNetBoxAppClock:014]: StatusFrameClock 13:43:00
truman-linbus/debug [D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
truman-linbus/debug [W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.

```

Many thanks again, and if I can get you any further information, please let me know.
Best
Jens

Fabian-Schmidt, 2023-11-02:
Thanks for the feedback.
Is this log recorded via wifi or serial logging?

Also can you please sent the log output of the `StatusFrameDevice` message. It should only be printed shortly after Esp boot up as it connects to the CP plus.

Mostly the CRC messages are no issue and highlight a timing issue. But the CP plus is repeating messages multiple times. So if Esp does not get everyone is not ideal but no big problem.

Cheers Fabian

gmanic, 2023-11-02:
The log is recorded wirelessly, via mqtt.

To get the serial output will take a couple days, as the esp is in my RV and I am still trying to figure a nice setup for the local (RV) homeassistant setup. I have two pi's alread in it (one for the Victron setup and one with OpenWrt for connectivity). Unfortunately, there's no "one-for-all" solution, yet. Maybe a pi4/8G with alpine-kvm/qemu plus three vm's might work.

But I'll try to get the serial output at the weekend, firing up the esp with my laptop connected.

Function test still to be done, as well. Good to know about the CRC messages.

Will keep you posted.
:)

gmanic, 2023-11-04:
Here's the startup scan. I deleted the CRC messages.
```
[D][wifi:428]: Starting scan...
[C][web_server:116]: Setting up web server...
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: 192.168.111.113:3232
[C][ota:101]:   Using Password.
[C][api:025]: Setting up Home Assistant API server...
[C][mqtt:037]: Setting up MQTT...
[D][binary_sensor:036]: 'CP Plus alive': Sending state OFF
[D][switch:055]: 'Activate Room Heater': Sending state OFF
[D][switch:055]: 'Activate Water Heater': Sending state OFF
[D][switch:055]: 'Activate Water Heater (enum)': Sending state OFF
[D][switch:055]: 'Active Timer': Sending state OFF
[W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[D][binary_sensor:036]: 'CP Plus alive': Sending state ON
[W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.FF.FF.FF (11)
[D][truma_inetbox.TrumaiNetBoxApp:148]: Requested read: Sending init
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FA.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.00.0A.01.9F (17)
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0C.0B.00.D4.02.00.01.00.50.00.00.04.01.02.57.0D.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[I][truma_inetbox.TrumaiNetBoxApp:303]: StatusFrameDevice
[D][truma_inetbox.TrumaiNetBoxApp:321]: StatusFrameDevice 1/2 - 4.01.02 0050.00 (57 0D)
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0C.0B.00.3F.02.01.01.00.40.03.0A.01.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[I][truma_inetbox.TrumaiNetBoxApp:303]: StatusFrameDevice
[D][truma_inetbox.TrumaiNetBoxApp:321]: StatusFrameDevice 2/2 - 1.02.00 0340.0A (00 00)
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.14.33.00.1F.00.00.00.00.00.00.00.00.00.00.01.01.0E.0B.18.0B.06.00.00.00.00.00.00.00 (41)
[I][truma_inetbox.TrumaiNetBoxApp:201]: StatusFrameHeater
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[D][main:484]: Message from CP Plus.
[D][binary_sensor:036]: 'Room Heater active': Sending state OFF
[D][binary_sensor:036]: 'Water Heater active': Sending state OFF
[D][binary_sensor:036]: 'Heater mode Gas': Sending state ON
[D][binary_sensor:036]: 'Heater mode Mix 1': Sending state OFF
[D][binary_sensor:036]: 'Heater mode Mix 2': Sending state OFF
[D][binary_sensor:036]: 'Heater mode Elec': Sending state OFF
[D][binary_sensor:036]: 'Heater has error': Sending state OFF
[D][climate:380]: 'Truma Room' - Sending state:
[D][climate:383]:   Mode: OFF
[D][climate:388]:   Fan Mode: OFF
[D][climate:403]:   Current Temperature: 11.00°C
[D][climate:409]:   Target Temperature: nan°C
[D][climate:380]: 'Truma Water' - Sending state:
[D][climate:383]:   Mode: OFF
[D][climate:403]:   Current Temperature: 10.00°C
[D][climate:409]:   Target Temperature: nan°C
[D][number:012]: 'Target Room Temperature': Sending state 0.000000
[D][number:012]: 'Target Water Temperature': Sending state 0.000000
[D][number:012]: 'electric power level': Sending state 0.000000
[D][sensor:094]: 'Current Room Temperature': Sending state 11.00000 °C with 1 decimals of accuracy
[D][sensor:094]: 'Current Water Temperature': Sending state 10.00000 °C with 0 decimals of accuracy
[D][sensor:094]: 'Target Room Temperature': Sending state nan °C with 0 decimals of accuracy
[D][sensor:094]: 'Target Water Temperature': Sending state nan °C with 0 decimals of accuracy
[D][sensor:094]: 'Heating mode': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:094]: 'electric power level': Sending state 0.00000 W with 0 decimals of accuracy
[D][sensor:094]: 'Energy mix': Sending state 1.00000  with 0 decimals of accuracy
[D][sensor:094]: 'Operating status': Sending state 6.00000  with 0 decimals of accuracy
[D][sensor:094]: 'Heater error code': Sending state 0.00000  with 0 decimals of accuracy
[W][component:214]: Component truma_inetbox took a long time for an operation (0.19 s).
[W][component:215]: Components should block for at most 20-30ms.
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.18.3D.00.53.00.00.00.00.00.00.00.00.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[I][truma_inetbox.TrumaiNetBoxApp:254]: StatusFrameTimer
[D][truma_inetbox.TrumaiNetBoxAppTimer:064]: StatusFrameTimer target_temp_room: nan target_temp_water: nan 00:00 -> 00:00  OFF
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[D][binary_sensor:036]: 'Timer active': Sending state OFF
[D][binary_sensor:036]: 'Timer Room Heater active': Sending state OFF
[D][binary_sensor:036]: 'Timer Water Heater active': Sending state OFF
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.17.00.0F.06.01.B4.0A.AA.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[I][truma_inetbox.TrumaiNetBoxApp:272]: StatusFrameConfig
[D][truma_inetbox.TrumaiNetBoxAppConfig:011]: StatusFrameConfig Offset: 0.0
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.15.00.51.11.25.00.01.01.00.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[... cut ...]
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[W][truma_inetbox.TrumaiNetBoxAppClock:031]: Invalid system time, not syncing to CP Plus.
[D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.15.00.50.11.26.00.01.01.00.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[I][truma_inetbox.TrumaiNetBoxApp:263]: StatusFrameClock
[D][truma_inetbox.TrumaiNetBoxAppClock:014]: StatusFrameClock 17:38:00
[D][truma_inetbox.LinBusProtocol:232]: Multi package response FB

```
I hope it helps.

gmanic, 2023-11-05:
First invocation via HA failed somehow. And worked as well. :)
My Truma setup is currently set on electric only (gas being worked on). As such, the Truma has been set to Electric1 (900W). That was (correctly) reported in HA.
When I set "room temp" to 18°, the setup switched to Gas (which failed with error "no gas" - and the error was correctly reported in HA). I did not systematically follow up on my next steps via HA (multiple times pressed the button for Electric1 (900W), to no avail.

After I switched to electric-only via the Truma-display, setting the room temperature worked and my Truma came to life, doing what it was said. Same goes for "off" or "room temp 5°".

Now it may take some further time on testing (I would like to be there to see/hear the response of the heater).

dougle03, 2024-02-17:
You should be able to get to the debug log from the ESP's web interface (IP address), No need to directly connect to it.

kaosphere34, 2025-01-17:
Hello ! I've the same message about de CRC : 0x00 LIN CRC error on SID. Have you fix the problem ? Thank you.