Fabian-Schmidt/esphome-truma_inetbox #21: Can't set HEATER_FAN_MODE to High/Boost

uhuck, 2023-12-29:
Hello Fabian.
I am excited about your development of the INet Box with ESPHome in the Home Assistant.
Thank you for that.

I have a motorhome with a Truma Combi 6E.
The connection is going very well so far. I can control everything.
Except that I can't set the FAN to the highest level. When I try this, the FAN mode strings back to Eco/Low or Medium mode.

I am sending you my ESPHome log below for examination:

INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/truma-inet.yaml...
INFO Updating https://github.com/Fabian-Schmidt/esphome-truma_inetbox.git@None
INFO Detected timezone 'Europe/Berlin'
INFO Starting log output from 192.168.1.208 using esphome API
INFO Successfully connected to truma-inet @ 192.168.1.208 in 0.151s
INFO Successful handshake with truma-inet @ 192.168.1.208 in 0.118s
[11:55:54][I][app:102]: ESPHome version 2023.12.5 compiled on Dec 25 2023, 20:36:59
[11:55:54][C][wifi:573]: WiFi:
[11:55:54][C][wifi:405]:   Local MAC: 0C:B8:15:F4:41:60
[11:55:54][C][wifi:410]:   SSID: 'KUBERT'[redacted]
[11:55:54][C][wifi:411]:   IP Address: 192.168.1.208
[11:55:54][C][wifi:413]:   BSSID: 00:1E:42:2B:E8:95[redacted]
[11:55:54][C][wifi:414]:   Hostname: 'truma-inet'
[11:55:54][C][wifi:416]:   Signal strength: -31 dB ▂▄▆█
[11:55:54][C][wifi:420]:   Channel: 11
[11:55:54][C][wifi:421]:   Subnet: 255.255.255.0
[11:55:54][C][wifi:422]:   Gateway: 192.168.1.1
[11:55:54][C][wifi:423]:   DNS1: 192.168.1.1
[11:55:54][C][wifi:424]:   DNS2: 0.0.0.0
[11:55:54][C][logger:443]: Logger:
[11:55:54][C][logger:444]:   Level: DEBUG
[11:55:54][C][logger:445]:   Log Baud Rate: 115200
[11:55:54][C][logger:447]:   Hardware UART: UART0
[11:55:54][C][uart.arduino_esp32:137]: UART Bus 1:
[11:55:54][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[11:55:55][C][uart.arduino_esp32:139]:   RX Pin: GPIO16
[11:55:55][C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[11:55:55][C][uart.arduino_esp32:143]:   Baud Rate: 9600 baud
[11:55:55][C][uart.arduino_esp32:144]:   Data Bits: 8
[11:55:55][C][uart.arduino_esp32:145]:   Parity: NONE
[11:55:55][C][uart.arduino_esp32:146]:   Stop bits: 2
[11:55:55][C][template.switch:068]: Template Switch 'Activate Room Heater'
[11:55:55][C][template.switch:091]:   Restore Mode: always OFF
[11:55:55][C][template.switch:068]: Template Switch 'Activate Water Heater'
[11:55:55][C][template.switch:091]:   Restore Mode: always OFF
[11:55:55][C][template.switch:091]:   Restore Mode: always OFF
[11:55:55][C][template.switch:057]:   Optimistic: NO
[11:55:55][C][template.switch:068]: Template Switch 'Active Timer'
[11:55:55][C][template.switch:091]:   Restore Mode: always OFF
[11:55:55][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[11:55:55][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[11:55:55][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[11:55:55][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus alive'
[11:55:55][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Room Heater active'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_ROOM'
[11:55:55][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Water Heater active'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_WATER'
[11:55:55][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater mode Gas'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_GAS'
[11:55:55][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater mode Mix 1'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_MIX_1'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_MIX_2'
[11:55:55][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater mode Elec'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_ELECTRICITY'
[11:55:55][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater has error'
[11:55:55][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_HAS_ERROR'
[11:55:55][C][truma_inetbox.timer_binary_sensor:029]: Truma Timer Binary Sensor 'Timer active'
[11:55:55][C][truma_inetbox.timer_binary_sensor:030]:   Type TIMER_ACTIVE
[11:55:55][C][truma_inetbox.timer_binary_sensor:029]: Truma Timer Binary Sensor 'Timer Room Heater active'
[... 222 lines cut ...]
[11:58:32][D][climate:396]: 'Truma Room' - Sending state:
[11:58:32][D][climate:399]:   Mode: HEAT
[11:58:32][D][climate:404]:   Fan Mode: MEDIUM
[11:58:32][D][climate:419]:   Current Temperature: 8.00°C
[11:58:33][D][climate:425]:   Target Temperature: 6.00°C
[11:58:33][D][climate:396]: 'Truma Water' - Sending state:
[11:58:33][D][climate:399]:   Mode: OFF
[11:58:33][D][climate:419]:   Current Temperature: 11.00°C
[11:58:33][D][climate:425]:   Target Temperature: nan°C
[11:58:33][D][number:012]: 'Target Room Temperature': Sending state 6.000000
[11:58:33][D][number:012]: 'Target Water Temperature': Sending state 0.000000
[11:58:33][D][number:012]: 'electric power level': Sending state 0.000000
[11:58:33][D][sensor:094]: 'Current Room Temperature': Sending state 8.00000 °C with 1 decimals of accuracy
[11:58:33][D][sensor:094]: 'Current Water Temperature': Sending state 11.00000 °C with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'Target Room Temperature': Sending state 6.00000 °C with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'Target Water Temperature': Sending state nan °C with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'Heating mode': Sending state 10.00000  with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'electric power level': Sending state 0.00000 W with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'Energy mix': Sending state 1.00000  with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'Operating status': Sending state 7.00000  with 0 decimals of accuracy
[11:58:33][D][sensor:094]: 'Heater error code': Sending state 0.00000  with 0 decimals of accuracy
[11:58:33][D][select:015]: 'Fan Mode': Sending state High (index 2)
[11:58:33][D][select:015]: 'Energy Mix': Sending state Gas (index 0)
[11:58:33][W][component:214]: Component truma_inetbox took a long time for an operation (0.24 s).
[11:58:33][W][component:215]: Components should block for at most 20-30ms.

uhuck, 2023-12-29:
Hello Fabian.
I've now figured out why you can't set the FAN to BOOST. The reason is that the FAN only has 2 levels and not 3.
There are only OFF, ECO and HOT.
I have a hard copy of my Truma app.
You can see the selection there.
If you have time, can you please change the code. Thank you!
[image]

uhuck, 2024-01-20:
Hi Fabian,
Könnest du mir bitte eine Rückmeldung geben, ob du eine Anpassung des FAN-MODE (Siehe meinen issue oben) eingeplant hast?
Falls du weitere Infos zur Steuerung meiner Truma Combi 6E brauchst, dann melde dich bitte.
Viele Grüße

Fabian-Schmidt, 2024-01-22:
Highest fan level (boost) only works when the current temperature is at least 10 degrees below target temperature and it will automatically switch to normal mode (eco/high) when done heating. This is enforced by the cp plus panel