Fabian-Schmidt/esphome-truma_inetbox #22: Alde no data

5of9, 2024-01-12:
Hi, I connected it to my Alde instead of the Inet Box. The only thing I get in the Esphome Device Logger is: [truma_inetbox.LinBusProtocol:226]	 Multi package request  BA.00.1F.00.1A.00.00.22.FF.FF.FF (11)
Do you know where to start?
Thank you

Fabian-Schmidt, 2024-01-12:
Hi,
I have no access to a Alde device so I am unable to develop the code for it.
Looks like the header is one byte different. I am unsure how much the Alde device is different from a Truma device.

Alde header:
`BA.00.1F.00.1A.00.00.22.FF.FF.FF (11)`
Truma header:
`BA.00.1F.00.1E.00.00.22.FF.FF.FF (11)`

5of9, 2024-01-13:
I would like to help you to get this done. What can I do for you? I have an ESP32 and an unused RPI here.

5of9, 2024-01-13:
Thats the record directly after switching on the original inetbox:
Time	level	Tag	Message
10:42:26	[D]	[binary_sensor:036]	
'CP Plus alive': Sending state ON
10:42:26	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BA.00.1F.00.1A.00.00.22.FF.FF.FF (11)
10:42:26	[D]	[truma_inetbox.TrumaiNetBoxApp:148]	
Requested read: Sending init
10:42:26	[D]	[truma_inetbox.LinBusProtocol:232]	
Multi package response FA.00.1F.00.1A.00.00.22.FF.FF.FF.54.01.00.0A.01.9F (17)
10:42:28	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BB.00.1F.00.14.00.00.22.FF.FF.FF.54.01.0E.0C.00.58.02.00.01.FF.DE.41.01.00.00.04.10.00.00.00 (31)
10:42:29	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BB.00.1F.00.14.00.00.22.FF.FF.FF.54.01.0E.0C.00.C9.02.01.01.FF.DE.41.30.03.00.02.6D.00.00.00 (31)
10:42:31	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BB.00.1F.00.12.00.00.22.FF.FF.FF.54.01.0C.55.00.D7.5E.0B.DE.03.C2.01.07.07.17.00.FF.3D (29)
10:42:32	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BB.00.1F.00.12.00.00.22.FF.FF.FF.54.01.0C.57.00.15.63.0B.DE.03.CE.04.07.07.00.00.FF.01 (29)
10:42:33	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BB.00.1F.00.10.00.00.22.FF.FF.FF.54.01.0A.15.00.48.0A.2A.09.FF.FF.FF.00.01.05.FF (27)
10:42:34	[D]	[internal_temperature:052]	
Ignoring invalid temperature (success=0, value=53.3)
10:42:34	[D]	[truma_inetbox.LinBusProtocol:226]	
Multi package request  BB.00.1F.00.22.00.00.22.FF.FF.FF.54.01.1C.51.00.92.01.00.5E.0B.FE.FF.00.0A.37.01.00.64.FF.FF.FF.82.63.0B.FE.FF.82.0A.00.1E.FF.FF.FF.FF (45)

5of9, 2024-01-21:
Ich wäre über jeden Tipp dankbar...

5of9, 2024-01-31:
Wenn du Hilfe benötigst oder ich was testen soll, jederzeit Bescheid geben!

Fabian-Schmidt, 2024-03-10:
Sorry for not answering. I was busy with other things :-|

I added an intitial detection for Alde device based on the information you provided commit e6fbece9882d20aa3dc36a11ce8832491ff9481d . Unfortunately I believe more changes are necessary to make it work.

Can you please test again and see what the next error / message are?

I am near Landshut this week and could drop by and take a look myself if this is feasible for you.

Cheers Fabian

5of9, 2024-03-10:
Hi, i will test it tomorrow evening👍🏼 you can find my Email adress if you Google my name and Landshut then we can maybe arrange a meeting. Good evening Florian Voit

Mic47114711, 2024-03-18:
Hallo Fabian, hallo Florian, ich habe eine Alde 3030 in meinem Wohnmobil eingebaut und möchte diese auch gern ansteuern. Habe mir gerade einen Womolin-TI-CI-LIN Controller besorgt und würde euch hier gern unterstützen! Welche Entwicklungsumgebung nutzt ihr? Vielleicht habe ihr Tipps, damit ich schnell arbeitsfähig werde.

Fabian-Schmidt, 2024-03-19:
Ist dein Alde paneel 'inet ready'?
Ich glaube alle nach 2018 hergestellten paneele sind es.

Mic47114711, 2024-03-19:
Hallo Fabian, ja ist es. Panel ist aus dem Jahr 2023. Ich habe gestern auch noch das Projekt mit esphome gebaut bekommen. Werde das versuchen heute Abend auf dem Controller zu installieren.

5of9, 2024-03-19:
Interessant, das ist neuer wie meins, die Bedienoberfläche schaut etwas anders aus, allerdings scheinen die Funktionen die gleichen zu sein. Wir sich zeigen ob die Datensätze die gleichen sind...
Mit esphome musst du auf jedem Fall Statusmeldungen empfangen, auch wenn die momentan noch nicht ausgewertet werden aber in Kürze gibt es erste Funktionen, wie ich Fabian kenne 😊

Mic47114711, 2024-03-20:
Konnte heute die Software auf dem Controller installieren. Ab Samstag bin in für zwei Wochen mit dem WoMo unterwegs, dann werde ich den Controller an die Alde anschliessen und schaue mal wie weit ich komme.

Fabian-Schmidt, 2024-03-23:
Hi,
Ich habe angefangen support für die ALDE experimentell einzubauen.

Ich habe es in einem seperaten branch eingebaut. Daher muss die external component referenz angepasst werden:

`source: github://Fabian-Schmidt/esphome-truma_inetbox@alde`

Bitte prüfen ob init funktioniert und welche Nachrichten kommen. 
Ich habe erstmal nur eine Komponente angepasst:

```
number:
  - platform: truma_inetbox
    name: "Target Room Temperature"
    type: TARGET_ROOM_TEMPERATURE
```

Alle anderen Komponenten (climate, number, sensor) kennen Alde noch nicht.

Als log bitte per USB serial log mit:

```
logger:
  baud_rate: 500000
  level: VERBOSE
```

Danke. Dann können wir schauen was funktioniert.

5of9, 2024-03-24:
Init funktioniert, esp wird als Inet Box Version ?.?.? erkannt, sonst funktioniert leider noch nichts. [truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.

ristoavila, 2024-08-05:
did you ever get this to work? I'm also getting Alde 3020 HE system and would be interested remote controlling it.

PrimaryInc, 2025-03-01:
Hi @Fabian-Schmidt ,

I have a Alde 3020 with Inet Ready 3020 panel in caravan.
I've been following different projects for Truma and trying to integrate esp home to my system and integrate it with Home Assistant. 

Question is if my setup and hardware can be of use to help create a working solution for controlling Alde.

Thanks!!

rolandtrieb, 2025-03-21:
> Init funktioniert, esp wird als Inet Box Version ?.?.? erkannt, sonst funktioniert leider noch nichts. [truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.

Hallo Florian, 
bist du mit der Alde denn weiter gekommen?
Danke und lg,
Roland

5of9, 2025-03-21:
Leider nicht dafür bin ich zu wenig Programmierer

rolandtrieb, 2025-04-01:
> Leider nicht dafür bin ich zu wenig Programmierer 
> -------- Ursprüngliche Nachricht --------Von: rolandtrieb ***@***.***> Datum: 21.03.25  09:12  (GMT+01:00) An: Fabian-Schmidt/esphome-truma_inetbox ***@***.***> Cc: Florian ***@***.***>, Author ***@***.***> Betreff: Re: [Fabian-Schmidt/esphome-truma_inetbox] Alde no data (Issue [#22](https://github.com/Fabian-Schmidt/esphome-truma_inetbox/issues/22)) 
> 
> Init funktioniert, esp wird als Inet Box Version ?.?.? erkannt, sonst funktioniert leider noch nichts. [truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
> 
> Hallo Florian,
> bist du mit der Alde denn weiter gekommen?
> Danke und lg,
> 
> 
> 
> rolandtrieb left a comment ([Fabian-Schmidt/esphome-truma_inetbox#22](https://github.com/Fabian-Schmidt/esphome-truma_inetbox/issues/22))
> 
> Init funktioniert, esp wird als Inet Box Version ?.?.? erkannt, sonst funktioniert leider noch nichts. [truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
> 
> Hallo Florian,
> bist du mit der Alde denn weiter gekommen?
> Danke und lg,
> Roland

Hallo Florian, danke für deine Rückmeldung! Hast du eine andere Lösung gefunden? Ich würde meine Alde gerne am Homeassistant ans laufen bekommen...
lg,
Roland

RiccardoLavagnino, 2025-04-04:
Ciao, I'm interested too in commanding and controlling Alde Heater from HA but it seems nothing complete available :(

HBK1, 2026-08-19:
This may help and is working:
https://github.com/holdestmade/esphome-alde-3020