Fabian-Schmidt/esphome-truma_inetbox #23: Operating Status 1 to 5

snipah, 2024-02-08:
Do we have any idea, what Operating Status 1 to 5 means? 5 is obviously "Heating Mode".

snipah, 2024-02-10:
Not needed anymore

closed 2024-02-10 by snipah

snipah, 2024-11-22:
> @snipah Have you found out what the numbers mean?

This is what I found out and what I'm using... 

`
- platform: template
    sensors:
      truma_status:
        friendly_name: "Truma Operating Status"
        value_template: >-
          {% set status_mapping = {
            0: "Aus",
            1: "Warnung",
            4: "Start/ Cool Down",
            5: "Heizen (5)",
            6: "Heizen (6)",
            7: "Heizen (7)"
          } %}
          {{ status_mapping[states('sensor.esp_poessl_operating_status')|int] if states('sensor.esp_poessl_operating_status')|int in status_mapping else 'Ungültig' }}`

7xy, 2025-01-24:
>   0: "Aus",
>   1: "Warnung",
>   4: "Start/ Cool Down",
>   5: "Heizen (5)", Status bei der Combi E wenn die Heizung an ist und kein Landstrom anliegt
>   6: "Heizen (6)", Status der Combi E, wenn die Heizung am CP Plus ausgeschaltet wurde und Landstrom anliegt
>   7: "Heizen (7)" Status bei der Combi E wenn die Heizung an ist und Landstrom anliegt

War zumindest meine Feststellung an meiner Combi 6E
[image]