Fabian-Schmidt/esphome-truma_inetbox #24: C3.00.01 H5.00.00

p0l0us, 2024-02-28:
Hi @Fabian-Schmidt 

Could you help me making this work with my iNet ready CP plus C3.00.01 H5.00.00 ?

I'm not sure if it's supposed to be supported by your project. I'm trying it, but I'm getting tons of CRC errors and "order no answers". See my config and the log bellow.

I'm using LIN converter board connected board Rx to Tx pin  GPIO17 and board Tx to Rx pin GPIO16.
Same board like here: https://github.com/mc0110/inetbox2mqtt/blob/main/doc/ELECTRIC.md 
I don't have any i2c RTC yet.

Thanks....

```
[12:04:32][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 28 2024, 12:03:40
[12:04:32][C][logger:447]: Logger:
[12:04:32][C][logger:448]:   Level: VERBOSE
[12:04:32][C][logger:449]:   Log Baud Rate: 115200
[12:04:32][C][logger:451]:   Hardware UART: UART0
[12:04:32][C][uart.arduino_esp32:108]: UART Bus 1:
[12:04:32][C][uart.arduino_esp32:109]:   TX Pin: GPIO17
[12:04:32][C][uart.arduino_esp32:110]:   RX Pin: GPIO16
[12:04:32][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[12:04:32][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[12:04:32][C][uart.arduino_esp32:115]:   Data Bits: 8
[12:04:32][C][uart.arduino_esp32:116]:   Parity: NONE
[12:04:32][C][uart.arduino_esp32:117]:   Stop bits: 2
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 16      00.0D.80.0B.94.0C.09.85.61 (9)  - SLAVE 
[12:04:32][C][truma_inetbox.room_climate:048]: truma_inetbox.room_climateTruma Room Climate 'Room'
[12:04:32][C][truma_inetbox.water_climate:019]: truma_inetbox.water_climateTruma Climate 'Water'
[12:04:32][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus alive'
[12:04:32][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater has error'
[12:04:32][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_HAS_ERROR'
[12:04:32][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[12:04:32][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[12:04:32][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[12:04:32][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[12:04:32][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[12:04:32][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[12:04:32][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 3C      FF.FF.FF.FF.FF.FF.FF.FF.00 (9)  - MASTER 
[12:04:32][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[12:04:32][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[12:04:32][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[12:04:32][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[12:04:32][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[12:04:32][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[12:04:32][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[12:04:32][C][truma_inetbox.heater_select:153]: Truma Heater Select 'Fan Mode'
[12:04:32][C][truma_inetbox.heater_select:153]:   Icon: 'mdi:thermometer'
[12:04:32][C][truma_inetbox.heater_select:154]:   Type 'HEATER_FAN_MODE'
[12:04:32][C][truma_inetbox.heater_select:155]:   Options are:
[12:04:32][V][truma_inetbox.LinBusListener:360]: PID 3D      order no answer
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 03      7C.0B.FF.FF.FF.FF.FF.FF.75 (9)  - SLAVE 
[12:04:32][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[12:04:32][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[12:04:32][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[12:04:32][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 04      AA.0A.FF.FF.FF.FF.FF.FF.86 (9)  - SLAVE 
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 06      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 16      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 16      00.0D.80.0B.94.0C.FF.84.6B (9)  - SLAVE 
[12:04:32][V][truma_inetbox.LinBusListener:360]: PID 17      order no answer
[12:04:32][V][truma_inetbox.LinBusListener:343]: RESPONSE 18 FE.FF.FF.FF.FF.FF.FF.FF (8)
[12:04:32][V][truma_inetbox.LinBusListener:392]: PID 00      FE.FF.FF.FF.FF.FF.FF.FF.28 (9)  - SLAVE 
[12:04:33][V][truma_inetbox.LinBusListener:392]: PID 03      7C.0B.FF.FF.FF.FF.FF.FF.75 (9)  - SLAVE 
[12:04:33][V][truma_inetbox.LinBusListener:392]: PID 04      AA.0A.FF.FF.FF.FF.FF.FF.86 (9)  - SLAVE 
[12:04:33][V][truma_inetbox.LinBusListener:392]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[12:04:33][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[12:04:33][V][truma_inetbox.LinBusListener:392]: PID 07      01.00.FF.FF.FF.FF.FF.FF.B7 (9)  - SLAVE 
[... 94 lines cut ...]
sensor:
  - platform: truma_inetbox
    name: "Current Room Temperature"
    type: CURRENT_ROOM_TEMPERATURE
  - platform: truma_inetbox
    name: "Current Water Temperature"
    type: CURRENT_WATER_TEMPERATURE

select:
  - platform: truma_inetbox
    name: "Fan Mode"
    type: HEATER_FAN_MODE_COMBI

truma_inetbox:
  uart_id: lin_uart_bus
  on_heater_message:
    then:
    - logger.log: "Message from CP Plus."

web_server:
  port: 80
  local: true
  version: 2
  include_internal: true
```

p0l0us, 2024-02-28:
After some experiments, I'm able to read some data, but not send any values to CP,

One of the important parts was solving issue above was resetting the truma CP plus in the menu.

Current issues:
- Target temperature is empty "[18:43:42][D][climate:425]:   Target Temperature: nan°C"
- I can't do any changes in the HA

Any ideas ?

_btw. In meantime I tried https://github.com/danielfett/inetbox.py. I was able to set temperature using mosquito commands._

<details>
<summary>Log output</summary>

```
[18:43:29][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 28 2024, 18:37:12
[18:43:29][C][uart.arduino_esp32:108]: UART Bus 1:
[18:43:29][C][uart.arduino_esp32:109]:   TX Pin: GPIO17
[18:43:29][C][uart.arduino_esp32:110]:   RX Pin: GPIO16
[18:43:29][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[18:43:29][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[18:43:29][C][uart.arduino_esp32:115]:   Data Bits: 8
[18:43:29][C][uart.arduino_esp32:116]:   Parity: NONE
[18:43:29][C][uart.arduino_esp32:117]:   Stop bits: 2
[18:43:29][C][truma_inetbox.room_climate:048]: truma_inetbox.room_climateTruma Room Climate 'Room'
[18:43:29][C][truma_inetbox.water_climate:019]: truma_inetbox.water_climateTruma Climate 'Water'
[18:43:29][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus alive'
[18:43:29][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater has error'
[18:43:29][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_HAS_ERROR'
[18:43:29][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[18:43:29][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[18:43:29][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[18:43:29][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[18:43:29][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[18:43:29][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[18:43:29][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[18:43:29][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[18:43:29][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[18:43:29][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[18:43:29][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[18:43:29][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[18:43:29][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[18:43:29][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[18:43:29][C][truma_inetbox.heater_select:153]: Truma Heater Select 'Fan Mode'
[18:43:29][C][truma_inetbox.heater_select:153]:   Icon: 'mdi:thermometer'
[18:43:29][C][truma_inetbox.heater_select:154]:   Type 'HEATER_FAN_MODE'
[18:43:29][C][truma_inetbox.heater_select:155]:   Options are:
[18:43:29][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[18:43:29][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[18:43:29][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[18:43:29][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[18:43:30][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.18.3D.00.53.00.00.00.00.00.00.00.00.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[18:43:30][I][truma_inetbox.TrumaiNetBoxApp:254]: StatusFrameTimer
[18:43:30][D][truma_inetbox.TrumaiNetBoxAppTimer:064]: StatusFrameTimer target_temp_room: nan target_temp_water: nan 00:00 -> 00:00  OFF
[18:43:30][D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[18:43:31][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.FF.FF.FF (11)
[18:43:31][W][truma_inetbox.TrumaiNetBoxApp:179]: Requested read: CP Plus asks for an update, but I have none.
[18:43:31][I][truma_inetbox.TrumaiNetBoxApp:254]: StatusFrameTimer
[18:43:31][D][truma_inetbox.TrumaiNetBoxAppTimer:064]: StatusFrameTimer target_temp_room: nan target_temp_water: nan 00:00 -> 00:00  OFF
[18:43:31][D][truma_inetbox.LinBusProtocol:232]: Multi package response FA
[18:43:34][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0C.0B.00.E0.02.00.01.00.50.00.00.03.00.01.4F.0C.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[18:43:34][I][truma_inetbox.TrumaiNetBoxApp:303]: StatusFrameDevice
[18:43:34][D][truma_inetbox.TrumaiNetBoxApp:321]: StatusFrameDevice 1/2 - 3.00.01 0050.00 (4F 0C)
[18:43:34][D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[18:43:35][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.FF.FF.FF (11)
[18:43:35][W][truma_inetbox.TrumaiNetBoxApp:179]: Requested read: CP Plus asks for an update, but I have none.
[18:43:35][I][truma_inetbox.TrumaiNetBoxApp:303]: StatusFrameDevice
[18:43:35][D][truma_inetbox.TrumaiNetBoxApp:321]: StatusFrameDevice 1/2 - 3.00.01 0050.00 (4F 0C)
[18:43:35][D][truma_inetbox.LinBusProtocol:232]: Multi package response FA
[... 203 lines cut ...]
    type: HEATER_HAS_ERROR

sensor:
  - platform: truma_inetbox
    name: "Current Room Temperature"
    type: CURRENT_ROOM_TEMPERATURE
  - platform: truma_inetbox
    name: "Current Water Temperature"
    type: CURRENT_WATER_TEMPERATURE

select:
  - platform: truma_inetbox
    name: "Fan Mode"
    type: HEATER_FAN_MODE_COMBI

truma_inetbox:
  uart_id: lin_uart_bus

web_server:
  port: 80
  local: true
  version: 2
  include_internal: true
```
</details>

p0l0us, 2024-03-03:
It seems working sometimes with config above - after the CP panel reset and loong waiting (5minutes?). But not everytime. I'm still setting "no data" and "CRC errors". 

<details>
<summary>Log output</summary>

```
[22:35:06][C][logger:447]: Logger:
[22:35:06][C][logger:448]:   Level: VERBOSE
[22:35:06][C][logger:449]:   Log Baud Rate: 115200
[22:35:06][C][logger:451]:   Hardware UART: UART0
[22:35:06][C][uart.arduino_esp32:108]: UART Bus 1:
[22:35:06][C][uart.arduino_esp32:109]:   TX Pin: GPIO17
[22:35:06][C][uart.arduino_esp32:110]:   RX Pin: GPIO16
[22:35:06][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[22:35:06][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[22:35:06][C][uart.arduino_esp32:115]:   Data Bits: 8
[22:35:06][C][uart.arduino_esp32:116]:   Parity: NONE
[22:35:06][C][uart.arduino_esp32:117]:   Stop bits: 2
[22:35:06][C][truma_inetbox.room_climate:048]: truma_inetbox.room_climateTruma Room Climate 'Room'
[22:35:06][C][truma_inetbox.water_climate:019]: truma_inetbox.water_climateTruma Climate 'Water'
[22:35:06][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus alive'
[22:35:06][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Heater has error'
[22:35:06][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_HAS_ERROR'
[22:35:06][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[22:35:06][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[22:35:06][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[22:35:06][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[22:35:06][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[22:35:06][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[22:35:06][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[22:35:06][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[22:35:06][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[22:35:06][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[22:35:06][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[22:35:06][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[22:35:06][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[22:35:06][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[22:35:06][V][truma_inetbox.LinBusListener:392]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[22:35:06][C][truma_inetbox.heater_select:153]: Truma Heater Select 'Fan Mode'
[22:35:06][C][truma_inetbox.heater_select:153]:   Icon: 'mdi:thermometer'
[22:35:06][C][truma_inetbox.heater_select:154]:   Type 'HEATER_FAN_MODE'
[22:35:06][C][truma_inetbox.heater_select:155]:   Options are:
[22:35:06][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[22:35:06][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[22:35:06][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[22:35:06][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[22:35:06][V][truma_inetbox.LinBusListener:392]: PID 00      03.10.29.BB.00.1F.00.1E.CA (9)  - MASTER 
[22:35:07][V][truma_inetbox.LinBusListener:392]: PID 04      3A.0C.FF.FF.FF.FF.FF.FF.F4 (9)  - SLAVE 
[22:35:07][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[22:35:07][V][truma_inetbox.LinBusListener:392]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[22:35:07][V][truma_inetbox.LinBusListener:360]: PID 17      order no answer
[22:35:07][W][truma_inetbox.LinBusListener:376]: LIN v1 CRC error
[22:35:07][V][truma_inetbox.LinBusListener:392]: PID 3C      FF.FF.FF.FF.FF.FF.FF.FF.20 (9)  - MASTER INVALID
[22:35:07][V][truma_inetbox.LinBusListener:392]: PID 03      72.0B.FF.FF.FF.FF.FF.FF.7F (9)  - SLAVE 
[22:35:07][V][truma_inetbox.LinBusListener:392]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[22:35:08][V][truma_inetbox.LinBusListener:392]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[22:35:08][V][truma_inetbox.LinBusListener:392]: PID 07      01.00.FF.FF.FF.FF.FF.FF.B7 (9)  - SLAVE 
[22:35:08][V][truma_inetbox.LinBusListener:343]: RESPONSE 18 FF.FF.FF.FF.FF.FF.FF.FF (8)
[22:35:08][V][truma_inetbox.LinBusListener:392]: PID 18      FF.FF.FF.FF.FF.FF.FF.FF.27 (9)  - SLAVE 
[22:35:08][V][truma_inetbox.LinBusListener:392]: PID 00      3A.0C.FF.FF.FF.FF.FF.FF.F4 (9)  - SLAVE 
[22:35:08][V][truma_inetbox.LinBusListener:392]: PID 00      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[22:35:09][V][truma_inetbox.LinBusListener:392]: PID 00      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[22:35:09][V][truma_inetbox.LinBusListener:392]: PID 04      3A.0C.FF.FF.FF.FF.FF.FF.F4 (9)  - SLAVE 
[22:35:09][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[22:35:09][V][truma_inetbox.LinBusListener:392]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[22:35:10][V][truma_inetbox.LinBusListener:360]: PID 3D      order no answer
[22:35:10][V][truma_inetbox.LinBusListener:360]: PID 3D      order no answer
[22:35:10][V][truma_inetbox.LinBusListener:360]: PID 3D      order no answer
[22:35:10][V][truma_inetbox.LinBusListener:392]: PID 03      72.0B.FF.FF.FF.FF.FF.FF.7F (9)  - SLAVE 
[22:35:10][V][truma_inetbox.LinBusListener:392]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[... 446 lines cut ...]
[22:36:06][V][truma_inetbox.LinBusListener:360]: PID 3D      order no answer
[22:36:06][V][truma_inetbox.LinBusListener:360]: PID 17      order no answer
[... cut ...]
[22:36:08][V][truma_inetbox.LinBusListener:360]: PID 3D      order no answer
[22:36:08][W][truma_inetbox.LinBusListener:373]: 0xEB LIN CRC error on SID.
[22:36:09][V][truma_inetbox.LinBusListener:392]: PID 04      3A.0C.FF.FF.FF.FF.FF.FF.F4 (9)  - SLAVE 
[22:36:09][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[22:36:09][V][truma_inetbox.LinBusListener:392]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[22:36:09][V][truma_inetbox.LinBusListener:360]: PID 17      order no answer
</details>

Fabian-Schmidt, 2024-03-10:
Part of the issue can be the logging.

ESP Home performs logging as a synchronous operation blocking the component while the log is written.
To minimize this blocking time decrease the log level and increase the log speed by setting the baud rate. Example:

```YAML
logger:
  baud_rate: 2000000
```

Can you please test with these changes again?

p0l0us, 2024-03-10:
@Fabian-Schmidt I'm not able to reproduce today.

Unfortunately you are not tagging releases so I'm not able to install older version and verify if it was fixed by your recent updates or if it's a random behavior. I will give it a try later again.

Fabian-Schmidt, 2024-03-12:
I am on purpose not tagging or creating releases to avoid bug reports in old versions. I believe the project is small enough that I can get away with it.
And for tech people they can always use the commit id as the target version.

p0l0us, 2024-03-13:
> And for tech people they can always use the commit id as the target version.

How can I use commit id ? apart of checking out manually and building myself ?

According to my findings, the config accepts only tag (or maybe branch name also). So in order to test older version I had to fork the project, tag it myself and then use following.

```
  - source: github://p0l0us/esphome-truma_inetbox@v0.61
```

IMO, Automated CI build number printed in initial log would be more convenient.

Fabian-Schmidt, 2024-04-08:
You are right the ESPHome does not support SHA in place of ref. I must be thinking of a different tool.

I might add a version number in the future. But currently I don't have enough time to do it properly.