Fabian-Schmidt/esphome-truma_inetbox #25: only "Multi package request" but no other Data

svoli, 2024-03-02:
Hello Fabian.
First of all, thank you very much for your work here!
But I have no idea what is wrong with my installation.
My hardware was immediately accepted by the Truma CP-Plus, but in the log I just keep seeing the same message "LinBusProtocol:226]: Multi package request". No further data is received and no data can be sent.
I don't think it's a hardware problem, because the connection is being established and data is being transmitted, just not enough. Or am I thinking wrong?

My Truma Hardware is:
C4.01.01 / 0050.00
H5.00.00 / 0301.01
T23.70.0 / IF00.01

And the Log from ESPHome:
INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/esp32truma.yaml...
INFO Starting log output from 10.30.0.53 using esphome API
INFO Successfully connected to esp32truma @ 10.30.0.53 in 0.067s
INFO Successful handshake with esp32truma @ 10.30.0.53 in 0.126s
[16:49:22][I][app:102]: ESPHome version 2024.2.1 compiled on Mar  2 2024, 15:36:50
[16:49:22][C][wifi:577]: WiFi:
[16:49:22][C][wifi:409]:   Local MAC: [redacted]
[16:49:22][C][wifi:414]:   SSID: [redacted]
[16:49:22][C][wifi:415]:   IP Address: 10.30.0.53
[16:49:22][C][wifi:417]:   BSSID: [redacted]
[16:49:22][C][wifi:418]:   Hostname: 'esp32truma'
[16:49:22][C][wifi:420]:   Signal strength: -44 dB ▂▄▆█
[16:49:22][C][wifi:424]:   Channel: 4
[16:49:22][C][wifi:425]:   Subnet: 255.255.255.0
[16:49:22][C][wifi:426]:   Gateway: 10.30.0.1
[16:49:22][C][wifi:427]:   DNS1: 10.30.0.1
[16:49:22][C][wifi:428]:   DNS2: 0.0.0.0
[16:49:22][C][logger:447]: Logger:
[16:49:22][C][uart.arduino_esp32:137]: UART Bus 1:
[16:49:22][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[16:49:22][C][uart.arduino_esp32:139]:   RX Pin: GPIO16
[16:49:22][C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[16:49:22][C][uart.arduino_esp32:143]:   Baud Rate: 9600 baud
[16:49:22][C][uart.arduino_esp32:144]:   Data Bits: 8
[16:49:22][C][uart.arduino_esp32:145]:   Parity: NONE
[16:49:22][C][uart.arduino_esp32:146]:   Stop bits: 2
[16:49:22][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[16:49:22][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[16:49:22][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[16:49:22][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[16:49:22][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus verbunden'
[16:49:23][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Truma ERROR'
[16:49:23][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_HAS_ERROR'
[16:49:23][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[16:49:23][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[16:49:23][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[16:49:23][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[16:49:23][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[16:49:23][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[16:49:23][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[16:49:23][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[16:49:23][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[16:49:23][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[16:49:23][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Room Temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[16:49:23][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[16:49:23][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[16:49:23][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[16:49:23][C][truma_inetbox.sensor:050]:   Type 'TARGET_ROOM_TEMPERATURE'
[16:49:23][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Water Temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[16:49:23][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[... 40 lines cut ...]
[16:49:23][C][ota:097]:   Address: esp32truma.local:3232
[16:49:23][C][ota:100]:   Using Password.
[16:49:23][C][ota:103]:   OTA version: 2.
[16:49:23][C][api:139]: API Server:
[16:49:23][C][api:140]:   Address: esp32truma.local:6053
[16:49:23][C][api:142]:   Using noise encryption: YES
[16:49:26][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:49:32][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:49:35][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.00.00.00.54.01.0A.15.00.47.10.31.00.01.01.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[16:49:37][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:49:42][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:49:47][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.00.00.00.54.01.14.33.00.05.FA.0A.01.00.00.00.00.00.00.00.01.01.D0.0C.69.0B.05.00.00.00.00.00.00.00 (41)
[16:49:49][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:49:53][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:49:58][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:50:00][D][select:062]: 'Fan Mode' - Setting
[16:50:00][D][select:115]: 'Fan Mode' - Set selected option to: High
[16:50:00][W][truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
[16:50:04][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)
[16:50:09][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)

Do you have any idea where the problem could be?
Thank you very much
Regards
Oliver

p0l0us, 2024-03-03:
Did you reset CP panel ? (Config - reset). It partially helped in my case. I also had to wait long time for syncing esp with panel. Maybe 5 or even 15 minutes (?). I'm not sure,  I gave up already and then, surprise, data appeared in HA and I was able to control the the heater.

Also you could try log level to verbose if you see anything else.

svoli, 2024-03-04:
Yes, I have reset it several times.  The ESP is also recognized by the CP panel. 
And it has been running for two days now and unfortunately I still only get the messages with "[D][truma_inetbox.LinBusProtocol:226]".
OK, I only had the log level set to debug so far. With Verbose I get more, but nothing that gets me any further:

INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/esp32truma.yaml...
INFO Starting log output from 10.30.0.53 using esphome API
INFO Successfully connected to esp32truma @ 10.30.0.53 in 0.176s
INFO Successful handshake with esp32truma @ 10.30.0.53 in 0.101s
[17:06:56][I][app:102]: ESPHome version 2024.2.1 compiled on Mar  4 2024, 16:57:39
[17:06:56][C][wifi:577]: WiFi:
[17:06:56][C][wifi:409]:   Local MAC: 94:3C:C6:38:A5:FC
[17:06:56][C][wifi:414]:   SSID: [redacted]
[17:06:56][C][wifi:415]:   IP Address: 10.30.0.53
[17:06:56][C][wifi:417]:   BSSID: [redacted]
[17:06:56][C][wifi:418]:   Hostname: 'esp32truma'
[17:06:56][C][wifi:420]:   Signal strength: -41 dB ▂▄▆█
[17:06:56][V][wifi:422]:   Priority: 0.0
[17:06:56][C][wifi:424]:   Channel: 4
[17:06:56][C][wifi:425]:   Subnet: 255.255.255.0
[17:06:56][C][wifi:426]:   Gateway: 10.30.0.1
[17:06:56][C][wifi:427]:   DNS1: 10.30.0.1
[17:06:56][C][wifi:428]:   DNS2: 0.0.0.0
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 04      FA.0A.FF.FF.FF.FF.FF.FF.F7 (9)  - SLAVE 
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 04      AA.0A.FF.FF.FF.FF.FF.FF.86 (9)  - SLAVE 
[17:06:56][C][logger:447]: Logger:
[17:06:56][C][logger:448]:   Level: VERBOSE
[17:06:56][C][logger:449]:   Log Baud Rate: 2000000
[17:06:56][C][logger:451]:   Hardware UART: UART0
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[17:06:56][C][uart.arduino_esp32:137]: UART Bus 1:
[17:06:56][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[17:06:56][C][uart.arduino_esp32:139]:   RX Pin: GPIO16
[17:06:56][C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[17:06:56][C][uart.arduino_esp32:143]:   Baud Rate: 9600 baud
[17:06:56][C][uart.arduino_esp32:144]:   Data Bits: 8
[17:06:56][C][uart.arduino_esp32:145]:   Parity: NONE
[17:06:56][C][uart.arduino_esp32:146]:   Stop bits: 2
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[17:06:56][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[17:06:56][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[17:06:56][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[17:06:56][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[17:06:56][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus verbunden'
[17:06:56][C][truma_inetbox.heater_binary_sensor:046]: Truma Heater Binary Sensor 'Truma ERROR'
[17:06:56][C][truma_inetbox.heater_binary_sensor:047]:   Type 'HEATER_HAS_ERROR'
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 07      E1.FE.FF.FF.FF.FF.FF.FF.D7 (9)  - SLAVE 
[17:06:56][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[17:06:56][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:06:56][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:06:56][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:06:56][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[17:06:56][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:06:56][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 16      00.05.04.0B.00.0B.3B.85.49 (9)  - SLAVE 
[17:06:56][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[17:06:56][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:06:56][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:06:56][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:06:56][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[17:06:56][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:06:56][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[17:06:56][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Room Temperature'
[17:06:56][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:06:56][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:06:56][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:06:56][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[17:06:56][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:06:56][C][truma_inetbox.sensor:050]:   Type 'TARGET_ROOM_TEMPERATURE'
[17:06:56][V][truma_inetbox.LinBusListener:392]: PID 3D      7F.06.B2.00.17.46.01.03.66 (9)  - MASTER 
[... 87 lines cut ...]
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 03      FA.0A.FF.FF.FF.FF.FF.FF.F7 (9)  - SLAVE 
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 04      AA.0A.FF.FF.FF.FF.FF.FF.86 (9)  - SLAVE 
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 07      E1.FE.FF.FF.FF.FF.FF.FF.D7 (9)  - SLAVE 
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 16      00.05.04.0B.00.0B.3B.85.49 (9)  - SLAVE 
[17:06:58][V][truma_inetbox.LinBusListener:392]: PID 3C      7F.06.B2.00.17.46.01.03.66 (9)  - MASTER 
[... cut ...]
[17:06:59][V][truma_inetbox.LinBusListener:392]: PID 3C      7F.06.B2.23.17.46.01.03.43 (9)  - MASTER 
[17:06:59][V][truma_inetbox.LinBusListener:343]: RESPONSE 18 FE.FF.FF.FF.FF.FF.FF.FF (8)
[17:06:59][V][truma_inetbox.LinBusListener:392]: PID 18      FE.FF.FF.FF.FF.FF.FF.FF.28 (9)  - SLAVE 
[17:06:59][V][truma_inetbox.LinBusListener:392]: PID 3C      03.10.0B.BA.00.1F.00.1E.E9 (9)  - MASTER 
[17:06:59][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BA.00.1F.00.1E.00.00.22.00.00.00 (11)

and so on.

p0l0us, 2024-03-06:
I'm also new here. Maybe you can compare config and logs with mine. It is mostly working.

https://github.com/Fabian-Schmidt/esphome-truma_inetbox/issues/24

Here ayou find my rough Truma & HA notes: https://github.com/p0l0us/PiHowTo/blob/main/RaspberryPI_4_HomeAssistant_Truma.md#esphome-approach

BTW. Enclosing logs and config  to following markdown will make your posts bit more readable:

````
<details>
<summary>Log output</summary>

```
The log in code block after 
empty line.
```
</details>
````

Fabian-Schmidt, 2024-03-10:
I see there is a difference in your Multi package request. So far I have only observed `BA.00.1F.00.1E.00.00.22.FF.FF.FF (11)`. While yours is `BA.00.1F.00.1E.00.00.22.00.00.00 (11)`. The last three bytes are `00` instead if `FF`.

I updated the code to ignore the last three bytes. Can you please test again?

svoli, 2024-03-10:
cool
Thank you Fabian! That has helped. Now everything works.
Thanks again!

closed 2024-03-10 by svoli