Fabian-Schmidt/esphome-truma_inetbox #26: Can`t connect to Truma

elojn, 2024-03-07:
Hi Fabian,
first of all, thanks for this project.
I tried to use it in my RV with Truma Combi 4, but it don´t work.

My Truma Hardware is:
C4.01.01 / 0050.00
H1.02.00 / 0300.02

 I tried with this basic program:

`esphome:
  name: "esphome-truma"

external_components:
  - source: github://Fabian-Schmidt/esphome-truma_inetbox
    components: ["truma_inetbox"]

esp32:
  board: mhetesp32devkit
  
# Enable logging
logger:
  level: Debug

uart:
  - id: lin_uart_bus
    tx_pin: 17
    rx_pin: 16  
    baud_rate: 9600
    stop_bits: 2

truma_inetbox:
  uart_id: lin_uart_bus

binary_sensor:
  - platform: truma_inetbox
    name: "CP Plus alive"
    type: CP_PLUS_CONNECTED

sensor:
  - platform: truma_inetbox
    name: "Current Room Temperature"
    type: CURRENT_ROOM_TEMPERATURE
  - platform: truma_inetbox
    name: "Current Water Temperature"
    type: CURRENT_WATER_TEMPERATURE
  - platform: truma_inetbox
    name: "Target Room Temperature"
    type: TARGET_ROOM_TEMPERATURE
  - platform: truma_inetbox
    name: "Target Water Temperature"
    type: TARGET_WATER_TEMPERATURE`

logging shows this while ESP ist connected to my laptop through USB (COM3):

INFO Successfully uploaded program.
INFO Starting log output from COM3 with baud rate 115200
[15:29:26][I][logger:355]: Log initialized
[15:29:26][I][app:029]: Running through setup()...
[15:29:26][C][uart.arduino_esp32:077]: Setting up UART...
[15:29:26][C][truma_inetbox.LinBusListener:042]: Setting up LIN BUS...
[15:29:26][I][app:062]: setup() finished successfully!
[15:29:26][D][binary_sensor:036]: 'CP Plus alive': Sending state OFF
[15:29:26][I][app:102]: ESPHome version 2023.12.5 compiled on Mar  7 2024, 15:28:39
[15:29:26][C][logger:443]: Logger:
[15:29:26][C][logger:444]:   Level: DEBUG
[15:29:26][C][logger:445]:   Log Baud Rate: 115200
[15:29:26][C][logger:447]:   Hardware UART: UART0
[15:29:26][C][uart.arduino_esp32:137]: UART Bus 1:
[15:29:26][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[... 100 lines cut ...]
[15:30:56][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:30:56][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:30:57][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:30:57][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:30:57][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:30:57][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:30:58][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:03][I][truma_inetbox.LinBusProtocol:153]: Assigned new SID 03
[15:31:03][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:04][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:04][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:04][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:05][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:05][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:05][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[15:31:05][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.

Do you have any advice ?

What makes me wondering is that the Reset at the CP is different as described . At my CP there is no flickering INIT... after I start the reset.

I tried the RESET procedure at the CP several times but nothing changed.

best regards
elojn

p0l0us, 2024-03-08:
Just idea. I also had issues running this project initially. So I tried https://github.com/danielfett/inetbox.py, just to ensure that my LIN adapter works with my Truma. When I knew, I can see data and send MQTT commands to the heater from my RPI OS command line, I came back to this project and HAOS.

Fabian-Schmidt, 2024-03-10:
Can you please share which lin bus controller you are using and how it is connected to the ESP32?

Also change logging to `VERBOSE` to see the lin messages.

elojn, 2024-03-10:
Hello Fabian,

thanks for your answer.
Here are pictures of my Hardware:
[image]
[image]

Connection to Truma:
[image]

Will try modified Loglevel asap.

Best regards

Fabian-Schmidt, 2024-03-10:
> Hello Fabian,
>
> thanks for your answer. Here are pictures of my Hardware: [...]

I am also using this LIN board and it is working.
Cables and config match. Apart from the color choice on one side pink and the other green for ground - That's dangerous.

elojn, 2024-03-11:
Today i tried with loglevel verbose.
Here is the log,  
**After Downloading and reboot:**
INFO Successfully uploaded program.
INFO Starting log output from COM3 with baud rate 115200
[13:12:15][     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[13:12:15][I][logger:355]: Log initialized
[13:12:15][I][app:029]: Running through setup()...
[13:12:15][V][app:030]: Sorting components by setup priority...
[13:12:15][C][uart.arduino_esp32:077]: Setting up UART...
[13:12:15][C][truma_inetbox.LinBusListener:042]: Setting up LIN BUS...
[13:12:15][I][app:062]: setup() finished successfully!
[13:12:15][D][binary_sensor:036]: 'CP Plus alive': Sending state OFF
[13:12:15][I][app:102]: ESPHome version 2023.12.5 compiled on Mar 11 2024, 13:11:24
[13:12:15][C][logger:443]: Logger:
[13:12:15][C][logger:444]:   Level: VERBOSE
[13:12:15][C][logger:445]:   Log Baud Rate: 115200
[13:12:15][C][logger:447]:   Hardware UART: UART0
[13:12:15][C][uart.arduino_esp32:137]: UART Bus 1:
[13:12:15][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[13:12:15][C][uart.arduino_esp32:139]:   RX Pin: GPIO16
[13:12:15][C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[13:12:15][C][uart.arduino_esp32:143]:   Baud Rate: 9600 baud
[13:12:15][C][uart.arduino_esp32:144]:   Data Bits: 8
[13:12:15][C][uart.arduino_esp32:145]:   Parity: NONE
[13:12:15][C][uart.arduino_esp32:146]:   Stop bits: 2
[13:12:15][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[13:12:15][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[13:12:15][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[13:12:15][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[13:12:15][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus alive'
[13:12:15][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[13:12:15][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[13:12:15][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[13:12:15][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[13:12:15][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[13:12:15][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[13:12:15][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[13:12:15][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[13:12:15][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[13:12:15][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[13:12:15][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Room Temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[13:12:15][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[13:12:15][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[13:12:15][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[13:12:15][C][truma_inetbox.sensor:050]:   Type 'TARGET_ROOM_TEMPERATURE'
[13:12:15][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Water Temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[13:12:15][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[13:12:15][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[13:12:15][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[13:12:15][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[13:12:15][C][truma_inetbox.sensor:050]:   Type 'TARGET_WATER_TEMPERATURE'
[13:12:22][  7034][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:23][  8641][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:23][  8693][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:23][  8745][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:23][  8797][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:24][  8817][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:24][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[13:12:24][  8869][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:24][  8923][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:24][  8976][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:12:24][  9028][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[... 128 lines cut ...]
[13:13:11][ 56282][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][ 56303][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[13:13:11][ 56355][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][ 56409][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][ 56461][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][ 56513][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][ 56565][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:11][W][truma_inetbox.LinBusListener:373]: 0x00 LIN CRC error on SID.
[13:13:11][ 56585][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[... cut ...]
[13:13:12][V][truma_inetbox.LinBusListener:392]: PID 3D      7F.06.B2.23.17.46.40.03.04 (9)  - MASTER
[13:13:12][ 56975][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:12][ 57027][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:12][ 57079][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:12][ 57131][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[13:13:12][ 57151][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.

Fabian-Schmidt, 2024-03-12:
There are some errors in the log:
`[E][truma_inetbox.LinBusListener:356]: PID 3D order - unable to send response`
This indicates that the CP Plus ask for information from the device and I tried to answer. But this answer did not made it on the wire, because If I did not observe the echo of my message.

Can you please increase the logger baud_rate to ensure the log output is not causing this issue:

```YAML
logger:
  baud_rate: 2000000
``` 

I assume the device is not in the list / index in the CP Plus device?

Can you please check that the cables make proper connection? Especially the TX wire.

elojn, 2024-03-16:
Thanks for reply.
will test with increased logger baudrate.

I tested the project although with different ESP32 board (NodeMcu WROOM32) with same issues.
with other LIN<-> TTL converter (same Type) although same result.
You are right the device ist not in the list/index in the CP Plus, even no flickering during RESET.

Other idea: could it help to test without logger ?  
In this case perhaps I found the device ind the list of the CP Plus?

elojn, 2024-03-16:
Next try:

suggested 
`logger:
  baud_rate: 2000000`
didn't work.

I tried without logger, but although didn't work.

with baudrate 38400 got this log data:

INFO Successfully uploaded program.
INFO Starting log output from COM3 with baud rate 38400
[17:21:42]\xb0n▬\xa9e§R\xd5▬\xd6M►$\x90B\x84\x90B§@"DD↑[I][logger:355]: Log initialized
[17:21:42][I][app:029]: Running through setup()...
[17:21:43][V][app:030]: Sorting components by setup priority...
[17:21:43][C][uart.arduino_esp32:077]: Setting up UART...
[17:21:43][C][truma_inetbox.LinBusListener:042]: Setting up LIN BUS...
[17:21:43][I][app:062]: setup() finished successfully!
[17:21:43][D][binary_sensor:036]: 'CP Plus alive': Sending state OFF
[17:21:43][I][app:102]: ESPHome version 2023.12.5 compiled on Mar 13 2024, 17:20:58
[17:21:43][C][logger:443]: Logger:
[17:21:43][C][logger:444]:   Level: VERBOSE
[17:21:43][C][logger:445]:   Log Baud Rate: 38400
[17:21:43][C][logger:447]:   Hardware UART: UART0
[17:21:43][C][uart.arduino_esp32:137]: UART Bus 1:
[17:21:43][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[17:21:43][C][uart.arduino_esp32:139]:   RX Pin: GPIO16
[17:21:43][C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[17:21:43][C][uart.arduino_esp32:143]:   Baud Rate: 9600 baud
[17:21:43][C][uart.arduino_esp32:144]:   Data Bits: 8
[17:21:43][C][uart.arduino_esp32:145]:   Parity: NONE
[17:21:43][C][uart.arduino_esp32:146]:   Stop bits: 2
[17:21:43][C][truma_inetbox.LinBusListener:032]: LinBusListener:
[17:21:43][C][truma_inetbox.LinBusListener:035]:   Update Interval: 0.5s
[17:21:43][C][truma_inetbox.LinBusListener:036]:   LIN checksum Version: 2
[17:21:43][C][truma_inetbox.LinBusListener:037]:   Observer mode: NO
[17:21:43][C][truma_inetbox.cpplus_binary_sensor:019]: Truma CP Plus Binary Sensor 'CP Plus alive'
[17:21:43][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Room Temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:21:43][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:21:43][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 1
[17:21:43][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:21:43][C][truma_inetbox.sensor:050]:   Type 'CURRENT_ROOM_TEMPERATURE'
[17:21:43][C][truma_inetbox.sensor:049]: Truma Sensor 'Current Water Temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:21:43][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:21:43][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[17:21:43][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:21:43][C][truma_inetbox.sensor:050]:   Type 'CURRENT_WATER_TEMPERATURE'
[17:21:43][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Room Temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:21:43][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:21:43][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[17:21:43][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:21:43][C][truma_inetbox.sensor:050]:   Type 'TARGET_ROOM_TEMPERATURE'
[17:21:43][C][truma_inetbox.sensor:049]: Truma Sensor 'Target Water Temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   Device Class: 'temperature'
[17:21:43][C][truma_inetbox.sensor:049]:   State Class: 'measurement'
[17:21:43][C][truma_inetbox.sensor:049]:   Unit of Measurement: '°C'
[17:21:43][C][truma_inetbox.sensor:049]:   Accuracy Decimals: 0
[17:21:43][C][truma_inetbox.sensor:049]:   Icon: 'mdi:thermometer'
[17:21:43][C][truma_inetbox.sensor:050]:   Type 'TARGET_WATER_TEMPERATURE'
[17:21:44][  2041][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:21:46][  3647][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:21:46][  3699][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:21:46][  3751][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[... 155 lines cut ...]
[17:22:14][ 31460][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:14][V][truma_inetbox.LinBusListener:349]: RESPONSE 3D 03.06.F2.00.00.00.00.00 (8)
[17:22:14][V][truma_inetbox.LinBusListener:398]: PID 3D      03.06.F2.00.00.00.00.00.04 (9)  - SLAVE
[17:22:14][ 31690][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:14][V][truma_inetbox.LinBusListener:398]: PID 3C      FF.FF.FF.FF.FF.FF.FF.FF.00 (9)  - MASTER
[17:22:14][ 31742][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:14][ 31850][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:14][V][truma_inetbox.LinBusListener:398]: PID 3C      FF.FF.FF.FF.FF.FF.FF.FF.00 (9)  - MASTER
[17:22:14][ 32007][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:14][I][truma_inetbox.LinBusProtocol:153]: Assigned new SID 03
[17:22:15][V][truma_inetbox.LinBusListener:398]: PID 3C      7F.06.B0.17.46.00.1F.03.4A (9)  - MASTER
[17:22:15][ 32059][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:15][V][truma_inetbox.LinBusListener:349]: RESPONSE 3D 03.01.F0.FF.FF.FF.FF.FF (8)
[17:22:15][V][truma_inetbox.LinBusListener:398]: PID 3D      03.01.F0.FF.FF.FF.FF.FF.0B (9)  - SLAVE [ 32113][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:15]
[... cut ...]
[17:22:15][ 32269][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:15][W][truma_inetbox.LinBusListener:379]: 0x00 LIN CRC error on SID.
[17:22:15][ 32289][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:15][ 32341][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:15][ 32445][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.
[17:22:15][ 32552][W][HardwareSerial.cpp:281] _uartEventTask(): UART1 RX break.

perhaps a step  to solve the problem ?