Fabian-Schmidt/esphome-truma_inetbox #27: Water target temperature 200°C

p0l0us, 2024-03-10:
Hi @Fabian-Schmidt 

When I set target water  temperature to 80°C, after a while target temperature 200°C is shown. Any idea what to try ?

[image]

<details>
<summary>Logs here:</summary>

<code>
[00:25:33][C][truma_inetbox.sensor:049]:   [truma_inetbox.LinBusListener:398]: PID 07      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 07      01.00.FF.FF.FF.FF.FF.FF.B7 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 16      04.15.30.0B.DC.0B.27.85.40 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:366]: PID 17      order no answer
[00:25:34][V][truma_inetbox.LinBusListener:349]: RESPONSE 18 FE.FF.FF.FF.FF.FF.FF.FF (8)
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 18      FE.FF.FF.FF.FF.FF.FF.FF.28 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 03      AA.0A.FF.FF.FF.FF.FF.FF.48 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 04      02.0D.FF.FF.FF.FF.FF.FF.2C (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 06      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 07      01.00.FF.FF.FF.FF.FF.FF.B7 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 16      04.15.30.0B.DD.0B.27.85.3F (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:366]: PID 17      order no answer
[00:25:34][V][truma_inetbox.LinBusListener:349]: RESPONSE 18 FE.FF.FF.FF.FF.FF.FF.FF (8)
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 18      FE.FF.FF.FF.FF.FF.FF.FF.28 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 3C      FF.FF.FF.FF.FF.FF.FF.FF.00 (9)  - MASTER 
[00:25:34][V][truma_inetbox.LinBusListener:366]: PID 3D      order no answer
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 03      AA.0A.FF.FF.FF.FF.FF.FF.48 (9)  - SLAVE 
[00:25:34][V][truma_inetbox.LinBusListener:398]: PID 04      02.0D.FF.FF.FF.FF.FF.FF.2C (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 05      01.FF.FF.FF.FF.FF.FF.FF.79 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 07      00.00.FF.FF.FF.FF.FF.FF.F9 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 07      01.00.FF.FF.FF.FF.FF.FF.B7 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 09      FF.FF.FF.FF.FF.FF.FF.FF.B6 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 17      04.15.30.0B.DD.0B.27.85.3F (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:366]: PID 17      order no answer
[00:25:35][V][truma_inetbox.LinBusListener:349]: RESPONSE 18 FE.FF.FF.FF.FF.FF.FF.FF (8)
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 18      FE.FF.FF.FF.FF.FF.FF.FF.28 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3D      03.05.B9.00.1F.00.00.FF.1F (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:349]: RESPONSE 3D 03.02.F9.00.FF.FF.FF.FF (8)
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3D      03.02.F9.00.FF.FF.FF.FF.01 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.10.29.BB.00.1F.00.1E.CA (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.21.00.00.22.FF.FF.FF.B9 (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 00      03.22.54.01.14.33.00.70.CD (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.23.2C.0B.01.00.00.00.A1 (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.24.7A.12.00.00.01.01.4A (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.25.E0.0B.30.0B.05.00.AB (9)  - MASTER 
[00:25:35][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.14.33.00.70.2C.0B.01.00.00.00.7A.12.00.00.01.01.E0.0B.30.0B.05.00.00.00.00.00.00.00 (41)
[00:25:35][I][truma_inetbox.TrumaiNetBoxApp:205]: StatusFrameHeater
[00:25:35][D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.26.00.00.00.00.00.00.D6 (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:349]: RESPONSE 3D 03.01.FB.FF.FF.FF.FF.FF (8)
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3D      03.01.FB.FF.FF.FF.FF.FF.00 (9)  - SLAVE 
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3C      03.05.B9.00.1F.00.01.FF.1E (9)  - MASTER 
[00:25:35][V][truma_inetbox.LinBusListener:349]: RESPONSE 3D 03.02.F9.00.FF.FF.FF.FF (8)
[00:25:35][V][truma_inetbox.LinBusListener:398]: PID 3D      03.02.F9.00.FF.FF.FF.FF.01 (9)  - SLAVE 
[00:25:36][V][truma_inetbox.LinBusListener:398]: PID 03      AA.0A.FF.FF.FF.FF.FF.FF.48 (9)  - SLAVE 
[00:25:36][D][climate:396]: 'Room' - Sending state:
[00:25:36][D][climate:399]:   Mode: HEAT
[00:25:36][D][climate:404]:   Fan Mode: LOW
[00:25:36][D][climate:419]:   Current Temperature: 13.40°C
[00:25:36][D][climate:425]:   Target Temperature: 13.00°C
[00:25:36][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[00:25:36][V][json:058]: Size after shrink 176 bytes
[00:25:36][D][climate:396]: 'Water' - Sending state:
[00:25:36][D][climate:399]:   Mode: HEAT
[00:25:36][D][climate:419]:   Current Temperature: 31.00°C
[00:25:36][D][climate:425]:   Target Temperature: 200.00°C
[00:25:36][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[00:25:36][V][json:058]: Size after shrink 160 bytes
[00:25:36][V][sensor:043]: 'Current Room Temperature': Received new state 13.399994
[00:25:36][D][sensor:094]: 'Current Room Temperature': Sending state 13.39999 °C with 1 decimals of accuracy
[00:25:36][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[00:25:36][V][json:058]: Size after shrink 92 bytes
[00:25:36][V][sensor:043]: 'Current Water Temperature': Received new state 31.000000
</code>

</details>

Fabian-Schmidt, 2024-03-12:
The Truma CP PLus has only three temperatures:
- Eco - 40°C
- High - 60°C
- Boost - 200°C

These are the technical numbers transmitted over the bus. So the Combi will obviously not heat to 200°C. But the CP Plus reports 200°C as the target temperature.

p0l0us, 2024-03-13:
Maybe this would be more consistent from user perspective ?

https://github.com/Fabian-Schmidt/esphome-truma_inetbox/pull/29

closed 2024-04-09 by p0l0us