Fabian-Schmidt/esphome-truma_inetbox #31: Compile error with Lilygo T-Dongle-S3

HBK1, 2024-04-10:
Dear Fabian!
I am trying to use a Lilygo T-Dongle-S3 but get errors when I compile the alde repository.
(Same error with truma repository)
Is this an ESPHome error or is it in truma-inetbox?

Config:

```
esphome:
  name: alde

esp32:
  board: esp32-s3-devkitc-1

external_components:
  - source: github://Fabian-Schmidt/esphome-truma_inetbox@alde
    refresh: 0s

ota:

web_server:
  port: 80

logger:
  baud_rate: 500000
  level: VERBOSE

uart:
  - id: lin_uart_bus
    rx_pin: 44
    tx_pin: 43
    baud_rate: 9600
    data_bits: 8
    parity: NONE
    stop_bits: 2

truma_inetbox:
  uart_id: lin_uart_bus

number:
  - platform: truma_inetbox
    name: "Target Room Temperature"
    type: TARGET_ROOM_TEMPERATURE

```

Compile:

```
INFO ESPHome 2024.3.2
INFO Reading configuration /config/esphome/alde.yaml...
INFO Updating https://github.com/Fabian-Schmidt/esphome-truma_inetbox.git@alde
INFO Generating C++ source...
INFO Compiling app...
Processing alde (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/LinBusListener.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/LinBusListener_esp32_arduino.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/LinBusListener_esp_idf.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/LinBusListener_rp2040.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/LinBusProtocol.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxApp.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppAirconAuto.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppAirconManual.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppAldeStatus.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppClock.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppConfig.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppHeater.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/TrumaiNetBoxAppTimer.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/helpers.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/number/TrumaAirconManualNumber.cpp.o
Compiling .pioenvs/alde/src/esphome/components/truma_inetbox/number/TrumaHeaterNumber.cpp.o
Compiling .pioenvs/alde/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o
Compiling .pioenvs/alde/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/alde/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
src/esphome/components/uart/uart_component_esp32_arduino.cpp: In member function 'virtual void esphome::uart::ESP32ArduinoUARTComponent::setup()':
src/esphome/components/uart/uart_component_esp32_arduino.cpp:90:25: error: cannot convert 'HWCDC*' to 'HardwareSerial*' in assignment
     this->hw_serial_ = &Serial;
                         ^~~~~~
*** [.pioenvs/alde/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o] Error 1
========================= [FAILED] Took 26.93 seconds =========================

```

HBK1, 2024-04-11:
If I omit logger or point to UART1 it compiles so the error seems to be the default logger uart USB_CDC that S3 uses. 

```
logger:
#  hardware_uart: USB_CDC
  hardware_uart: UART1
  baud_rate: 50000
  level: VERY_VERBOSE
```

HBK1, 2024-04-11:
If I comment out the Truma parts it compiles with USB_CDC without problem.
So it looks like there may be something in your files that needs updating to support USB_CDC for logging.

```
esphome:
  name: alde

esp32:
  board: esp32-s3-devkitc-1

#external_components:
#  - source: github://Fabian-Schmidt/esphome-truma_inetbox@alde
#    refresh: 0s

ota:

web_server:
  port: 80

logger:
  hardware_uart: USB_CDC
  baud_rate: 50000
  level: VERY_VERBOSE

uart:
  - id: lin_uart_bus
    rx_pin: 44
    tx_pin: 43
    baud_rate: 9600
    data_bits: 8
    parity: NONE
    stop_bits: 2

#truma_inetbox:
#  uart_id: lin_uart_bus

#number:
#  - platform: truma_inetbox
#    name: "Target Room Temperature"
#    type: TARGET_ROOM_TEMPERATURE

```

HBK1, 2024-04-11:
When I compare uart_component_esp32_arduino.cpp and uart_component_esp32_arduino.h with the esphome repository it differs  and the esphome repo seems to handle the logging over USB_CDC.
I am not a coder so I may be wrong.

Fabian-Schmidt, 2024-04-14:
Can you please test the following external component:

```yaml
external_components:
  - source: github://Fabian-Schmidt/esphome-truma_inetbox
    components: ["truma_inetbox"]
```

The uart component was needed in the first release and the changes have been made upstream in esphome.
I require it sometimes for local testing things and might remove it in the future.

HBK1, 2024-04-14:
Thanks Fabian, now it compiles. Will now try if I can get some logs from my Alde 3020.

closed 2024-04-14 by HBK1