Fabian-Schmidt/esphome-truma_inetbox #32: How can I get the actual time value of the Truma Heating system

Karatzie, 2024-04-18:
Hi Fabian,

thank you for your great ESPHome-Truma Inetbox connectivity.

I use this connectivity on my vacation last week.
It was very cold and I had managed my Truma heating with Home Assistant.
I would like to get the actual time value from the heating in Home Assistant.
I can see the actual truma time on the ESPHome log.
In this case, I can compare both time values from Home Assistant with the time of the truma heating.
If I observe a different time, I can activate the truma_inetbox.clock.set action and the truma heating get the correct time.
We have in Germany two time shifts to summer-time and normal-time. This time shift should be observed by Home Assistant and the commant truma_inetbox.clock.set should set the new time of the truma heating.

Please add the actual truma heating time to an entity in your connectivity.

Thanks a lot and best regards

Andreas (karatzie)

Karatzie, 2024-07-29:
Hi Fabian,

have you read this improvement?

Best regards
Andreas (karatzie)

Lemmork, 2025-02-16:
@Karatzie could you share your setup, how you set the clock?
I created a button, but the clock is not set:
button:
  - platform: template
    name: "Set CP-Plus Time"
    on_press:
      - truma_inetbox.clock.set

Karatzie, 2025-02-16:
@Lemmork 
Here is my code to set the clock.
It is not working every time. If the time from Home Assistant not valid, the time is not set.
In this case I reboot the ESP32.
I use the time from Home Assistant.
You can see the running action in the log with "Synchronized home assistant system clock".
The timesync from Home Assistant is updating every 15 minutes.

`time:
#  - platform: ds1307
#    update_interval: never
#    id: esptime
# Example configuration entry
#time:
  - platform: homeassistant
    id: esptime
    on_time_sync:
      then:
        - logger.log:
            format: "Synchronized home assistant system clock %s"
            args: [ 'id(esptime).now()']

uart:
  - id: lin_uart_bus
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600
    data_bits: 8
    parity: NONE
    stop_bits: 2

truma_inetbox:
  uart_id: lin_uart_bus
  time_id: esptime