Fabian-Schmidt/esphome-truma_inetbox #34: sometimes long processing time and CRC error

rzaenger, 2024-04-30:
Dear Fabian,

First of all, thank you very much for your realization of this project, I am really very satisfied with the result, which can be realized -via Home Assistant- in the CamperVan! 

I noticed these two highlights in my log files:

1. the sometimes long processing time 

2. sporadic CRC errors

What do you think could be the cause?

**[10:22:07][W][component:237]: Component truma_inetbox took a long time for an operation (228 ms).
[10:22:07][W][component:238]: Components should block for at most 30 ms.**
[10:22:09][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.18.3D.00.53.00.00.00.00.00.00.00.00.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[10:22:09][I][truma_inetbox.TrumaiNetBoxApp:258]: StatusFrameTimer
[10:22:09][D][truma_inetbox.TrumaiNetBoxAppTimer:064]: StatusFrameTimer target_temp_room: nan target_temp_water: nan 00:00 -> 00:00  OFF
[10:22:09][D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[10:22:09][D][binary_sensor:036]: 'Timer active': Sending state OFF
[10:22:09][D][binary_sensor:036]: 'Timer Room Heater active': Sending state OFF
[10:22:09][D][binary_sensor:036]: 'Timer Water Heater active': Sending state OFF
[10:22:11][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.17.00.0F.06.01.B4.0A.AA.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[10:22:11][I][truma_inetbox.TrumaiNetBoxApp:276]: StatusFrameConfig
[10:22:11][D][truma_inetbox.TrumaiNetBoxAppConfig:011]: StatusFrameConfig Offset: 0.0
[10:22:11][D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
[10:22:13][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.15.00.2D.0A.16.3B.01.01.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)
[10:22:13][I][truma_inetbox.TrumaiNetBoxApp:267]: StatusFrameClock
[10:22:13][D][truma_inetbox.TrumaiNetBoxAppClock:014]: StatusFrameClock 10:22:59
[10:22:13][D][truma_inetbox.LinBusProtocol:232]: Multi package response FB
**[10:22:13][W][truma_inetbox.LinBusListener:382]: LIN v1 CRC error**
[10:22:15][D][truma_inetbox.LinBusProtocol:226]: Multi package request  BB.00.1F.00.1E.00.00.22.FF.FF.FF.54.01.0A.15.00.67.0A.17.00.01.01.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (41)

Best regards 
Roland

danistst, 2025-01-03:
Hi Roland

I'm having the same issue. Did you manage to solve it?

Cheers
Dani

rzaenger, 2025-01-03:
Hi Dani,

Yes, a little bit. I used ChatGPT to optimize the code and I have reduced the distance to the Raspi with Home Assistant in my CamperVan.

BR
Roland

danistst, 2025-01-03:
Interesting! Does it work better now? 

Do you mean the distance between the RPi with HA, your Wi-Fi router and the ESP with the Truma_inetbox? Or the LIN cable between the ESP and the Truma heater?

And could you elaborate on the optimization of the code?

Cheers
Dani

rzaenger, 2025-01-05:
Hi, I mean the distance between the ESP32 and the router. I just had the YAML code optimized at ChatGPT by giving a prompt that looks like this: please optimize the following code to generate as little latency as possible:
[image]