Fabian-Schmidt/esphome-truma_inetbox #37: Support for Truma Aventa compact

iBobik, 2024-07-02:
Hi, I just bought Aventa and would like to integrate it.

Can I help you with remote debugging and sponsor your work?

iBobik, 2024-07-08:
Tried this config:
```
substitutions:
  id_name: cooling
  human_name: Cooling

<<: !include network_base.yml

esphome:
  name: $id_name
  friendly_name: $human_name
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

external_components:
  - source: github://Fabian-Schmidt/esphome-truma_inetbox
    components: ["truma_inetbox"]

uart:
  - id: lin_uart_bus
    baud_rate: 9600
    stop_bits: 2
    rx_pin: GPIO6
    tx_pin: GPIO5

truma_inetbox:
  uart_id: lin_uart_bus

binary_sensor:
  - platform: truma_inetbox
    name: "CP Plus alive"
    type: CP_PLUS_CONNECTED

  - platform: truma_inetbox
    name: "Timer active"
    type: TIMER_ACTIVE

  - platform: truma_inetbox
    name: "Timer room"
    type: TIMER_ROOM

sensor:
  - platform: truma_inetbox
    name: "Current Room Temperature"
    type: CURRENT_ROOM_TEMPERATURE

  - platform: truma_inetbox
    name: "Target Room Temperature"
    type: TARGET_ROOM_TEMPERATURE

  - platform: truma_inetbox
    name: "Operating Status"
    type: OPERATING_STATUS

climate:
  - platform: truma_inetbox
    name: "Truma Room"
    type: ROOM

number:
  - platform: truma_inetbox
    name: "Target Room Temperature"
    type: TARGET_ROOM_TEMPERATURE

  - platform: truma_inetbox
    name: "Electric Power Level"
    type: ELECTRIC_POWER_LEVEL

  - platform: truma_inetbox
    name: "Aircon Manual Temperature"
    type: AIRCON_MANUAL_TEMPERATURE

select:
  - platform: truma_inetbox
    name: "Fan Mode"
    type: HEATER_FAN_MODE_COMBI

```

Then started the controller and after a minute turned on power for Aventa and using remote started air conditioning. Added device to the Home Assistant and tried to change values, but none worked:
```
[16:31:07][D][number:054]: 'Aircon Manual Temperature' - Setting number value
[16:31:07][D][number:113]:   New number value: 31.000000
[16:31:07][W][truma_inetbox.TrumaiNetBoxAppAirconManual:055]: Cannot update Truma.
[16:32:31][D][climate:011]: 'Truma Room' - Setting
[16:32:31][D][climate:024]:   Fan: OFF
[16:32:31][W][truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
[16:32:43][D][climate:011]: 'Truma Room' - Setting
[16:32:43][D][climate:015]:   Mode: HEAT
[16:33:00][D][climate:011]: 'Truma Room' - Setting
[16:33:00][D][climate:024]:   Fan: LOW
[16:33:00][W][truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
[16:33:25][D][number:054]: 'Target Room Temperature' - Setting number value
[16:33:25][D][number:113]:   New number value: 30.000000
[16:33:25][W][truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
[16:33:37][D][select:062]: 'Fan Mode' - Setting
[16:33:37][D][select:115]: 'Fan Mode' - Set selected option to: Eco
[16:33:37][W][truma_inetbox.TrumaiNetBoxAppHeater:059]: Cannot update Truma.
[16:33:45][D][number:054]: 'Electric Power Level' - Setting number value
[16:33:45][D][number:113]:   New number value: 0.000000
[16:33:45][W][truma_inetbox.TrumaiNetBoxAppHeater:141]: Cannot update Truma.
```

smai86, 2025-03-05:
Hey @iBobik,
do you get it work with the aventa compact?
I‘m actually only using a truma combi 4 and thinking of buying an aventa compact this summer.
It would be nice if you could share your experience.

Thanks a lot!

iBobik, 2025-03-05:
> Hey [@iBobik](https://github.com/iBobik), do you get it work with the aventa compact? I‘m actually only using a truma combi 4 and thinking of buying an aventa compact this summer. It would be nice if you could share your experience.
> 
> Thanks a lot!

No, Aventa cannot be controlled by this.

smai86, 2025-03-05:
> > Hey [@iBobik](https://github.com/iBobik), do you get it work with the aventa compact? I‘m actually only using a truma combi 4 and thinking of buying an aventa compact this summer. It would be nice if you could share your experience.
> > Thanks a lot!
> 
> No, Aventa cannot be controlled by this.

:( thats bad… have you found an other solution to control the aventa with home assistant?

iBobik, 2025-03-05:
> have you found an other solution to control the aventa with home assistant?

There is also https://github.com/mc0110/inetbox2mqtt but I was not able to make it work even for the heating.

RiccardoLavagnino, 2025-03-15:
i'm investigating another solution: to decode Truma App (developed for android with  Cordova) and rebuilt with MQTT communication to HA. I already have the truma inetbox and it could be the best solution for controlling in HA both Alde and Aventa...but it's not easy :(

iBobik, 2025-03-15:
> i'm investigating another solution: to decode Truma App (developed for android with Cordova) and rebuilt with MQTT communication to HA. I already have the truma inetbox and it could be the best solution for controlling in HA both Alde and Aventa...but it's not easy :(

How exactly? Sniff packets it will send to the LIN?

iBobik, 2025-03-15:
I had this idea:

Try to search specific command in the sources of the mc0110/inetbox2mqtt and then port them to this project. If they implemented it correctly, it should work.

RiccardoLavagnino, 2025-03-15:
No, acting like a Truma app connected to original Inetbox with the additional possibility to communicate via MQTT to HA. In other words in case of success, I’ll leave an old android smartphone in the camper…but I’m struggling in rebuilding the decoder app (without any modifications for now).

iBobik, 2025-03-15:
> No, acting like a Truma app connected to original Inetbox with the additional possibility to communicate via MQTT to HA. In other words in case of success, I’ll leave an old android smartphone in the camper…but I’m struggling in rebuilding the decoder app (without any modifications for now).

I had inetbox borrowed and I remember it communicated only by bluetooth, so one more point of failure. I also remember there were not many sensors and switches what this project have.

RiccardoLavagnino, 2025-03-16:
You’re right but the entire LIN protocol from inetbox to Alde heater is quite not discovered…while having the entire arc code of the app->inetbox app it is just a matter of implementing