Fabian-Schmidt/esphome-truma_inetbox #39: Usage without esp32

temuccio, 2024-07-30:
HI,
I am interested in this project as I see that the Truma Saphir AC that I own has been implemented.
Can I use this project with a LIN->UART->FTDI R232 UART converter, without using esp32?

Thanks

savikko, 2024-07-31:
Short answer: no.

Longer: This is component for ESPHome and ESPHome requires some compatible board (see https://esphome.io/index.html#supported-microcontrollers).
I think project is mostly used with ESP32. ESP8266 might not be powerful enough and RP/BK/RTL are still considered somewhat "experimental" within ESPHome.

But no, you cannot run this project on your Mac/Win/Linux device using LIN/UART/FTDI converter. Additionally, I dont think there is any project which has that kind of implementation.

Easiest way to proceed is to use some readymade ESP32 appliance which connects to your Truma.
(eg https://shop.smartavan.com/products/smartavan-heater-controller - disclaimer: I am affiliated with it as its seller/developer.)

temuccio, 2024-07-31:
Hi @savikko, 
thanks for replay.

I'm not sure if I will be able to interface my Truma Saphir AC. I previously had the X Panel, then I purchased the CP Plus, but trying with danielfett/inetbox.py didn't work.

I read among the issues of this project that the Truma Saphir AC should be interfaced, so I wanted to give it a try.

I'm trying to install the ESPHome docker to see if I can manage without additional hardware.

I will keep you updated.

savikko, 2024-07-31:
> I'm trying to install the ESPHome docker to see if I can manage without additional hardware.

Note that ESPHome docker image is not runtime, it is the build platform.

There is possibility to run ESPHome on Mac/Linux environments (https://esphome.io/components/host), but I doubt this project will run with that.

closed 2024-08-07 by temuccio