Fabian-Schmidt/esphome-truma_inetbox #40: FR: Electricity connected binary sensor

savikko, 2024-08-04:
It seems that there is not currently binary sensor to see if outside electricity 230VAC is connected.

Although, it seems that OPERATING_STATUS changes from 5 to 7 when electricity is connected but it also seems that this status has other functionalities so this cannot be used as a binary sensor here.

So, this sensor would be only to get info that is there outside electricity connected.

Will update this issue to provide the source for this info (I have found this somewhere, but needs little bit searching).

savikko, 2024-08-05:
Ah, operating status is actually binary, so quicky hack to get that data is to parse bits from that value like this:

```
binary_sensor:
  - platform: template
    name: "Supply Electricity"
    lambda: |-
      return ((int) id(operating_status).state & 0b00000010) != 0;
  - platform: template
    name: "Room Demand"
    lambda: |-
      return ((int) id(operating_status).state & 0b00001000) != 0;
  - platform: template
    name: "Water Demand"
    lambda: |-
      return ((int) id(operating_status).state & 0b00010000) != 0;

sensor:
  - platform: truma_inetbox
    name: "Operating status"
    id: operating_status
    type: OPERATING_STATUS
```

7xy, 2025-01-24:
> [@savikko](https://github.com/savikko) Could you please provide more guidance how this is used? 

You have to integrate the new binary sensor in your trum.yaml under ESPHome and add the line "id: operating_status" to the sensor for operating status. Then save and install the trum.yaml file!

After that you will have a new sensor named "Supply Electricity" for your Truma.