Fabian-Schmidt/esphome-truma_inetbox #48: Current Room and Water temperature only when heater is active?

eRoyaner, 2025-01-17:
Hi Fabian, Thank you for your great work here.

I already got everything running on my esp32 together with the homeassistant integration.
I only have one Problem that the Current Room and Water temperature is only displayed correctly when I activate the heater.
If heater is deactivated but still alive then I get only 0°C value.

[image]

Can you tell me what my problem is? 

Regards,
Fabian ;)

7xy, 2025-01-22:
> Hi Fabian, Thank you for your great work here.
>
> I already got everything running on my esp32 together with the homeassistant integration. I only have one Problem that the Current Room and Water [...]

Wie hast Du die Darstellung der Heizungssteuerung umgesetzt? Kannst Du netterweise Deine Yaml-Datei der card hier einstellen! Sieht extrem gut aus!

bustinex, 2025-02-01:
I have the same problem. I think that the truma sends the temperature data only if it is turned on.

Do you know how to control ONLY the ventilation? (cp-plus 0-10 fan)