Fabian-Schmidt/esphome-truma_inetbox #53: ALDE 3020HE Integration

rolandtrieb, 2025-03-27:
Hallo Fabian!
Erstmal vielen Dank für deine großartige Arbeit!

Aktuell habe ich den WomoLin Controller mit deinem Alde Branch als Inet Box Emulation am laufen. Ein paar Werte kommen bereits retour und die Bus Kommunikation scheint zu funktionieren:

[image]

Kannst du helfen um die Heizung vollständig in deine Lösung zu integrieren?

Würde mich freuen!
Vielen Dank vorab und lg,
Roland

RiccardoLavagnino, 2025-06-16:
Do you have any progress on it? I’ve an Alde and I’m struggling find the way for getting the integration

HBK1, 2026-08-19:
I am using below in ESPHome and it works ok.
https://github.com/holdestmade/esphome-alde-3020