Fabian-Schmidt/esphome-truma_inetbox #55: Truma Combi 6DE works! But one Question -> Vent?

Doka85, 2025-04-07:
Hello everyone,

First of all, many thanks to **Fabian Schmidt** for the great ESPHome implementation – and of course also to **Daniel Fett**, **muccc**, and all others who helped lay the foundation for this amazing project.

Today, I successfully integrated a **Truma Combi 6DE** into my setup, and everything seems to work flawlessly so far. I was able to toggle through the modes (`mix1`, `mix2`, `gas`/`diesel`, `elec`), and my Volkswagen Grand California’s camper unit correctly showed the corresponding status.

Due to a missing 230V connection, I couldn’t fully test electric heating yet, but I’ll try this over the weekend. Still, I’m confident everything will work as expected based on the feedback shown on the camper control panel.

I’m currently still fine-tuning my YAML and would be happy to share the final working version afterward.

---

However, I have one open question:

The **CP+ panel** allows a **ventilation-only mode** (fan only, no heating) with **10 adjustable fan speed levels**. I read in Daniel Fett’s repo (he wrote that 2022, an i found no Update) that this is *currently not implemented*. Am I correct in assuming that this feature is also *not yet supported in this ESPHome version*?

In his blog post, **Daniel Schleeh** includes the following code snippet in his YAML, which (according to him) seems related to fan functionality – but I believe this is more intended for **air conditioning units**, correct?

```yaml
  - platform: truma_inetbox
    name: "Aircon Temperature"
    type: AIRCON_MANUAL_TEMPERATURE`
```

Thank you again for this amazing project – and if there's any way I can help with logs or testing to support vent-only mode implementation, I’d be happy to contribute.

Best regards,
Doka

photogeen, 2025-04-26:
Hi, I have the Truma 6E and it also has the Vent-only mode. +1
How can I support? Would love to test.

Eugen

Stiles96, 2025-05-11:
I like the vent mode also. +1

Stefan132-new, 2026-05-11:
@Doka85 @photogeen 
have you found a solution for using just the vent without heating? Would be great together with my maxxfan roof vent :-)

@Fabian-Schmidt thanks for you efforts. Works with TI-CI-LIN Interface v2.1 and esp32 with esphome in HA