Fabian-Schmidt/esphome-truma_inetbox #63: Is this project now abandoned?

darrenjackson72, 2025-07-17:
I'm sure most are noticing that there are no responses to issues any more?

santiniuk, 2025-11-05:
Hi,

Trying to get a working version and tried all of the above to compile but still errors.
Does anyone have a solution to compile in ESPhome please?
version 2025.10.4

Thanks

darrenjackson72, 2025-11-05:
Awesome.. If you are running the WOMOLIN kit you are good to go...

darrenjackson72, 2025-11-05:
# Complete Optimized Truma ESPHome Configuration
# DS1307/I2C issues fixed - using SNTP for reliable time sync

esphome:
  name: "truma"
  friendly_name: "Truma Inetbox"
  min_version: 2024.11.0
  name_add_mac_suffix: false

  on_boot:
    priority: -100
    then:
      - logger.log: "Truma system starting with SNTP time sync"
      - delay: 5s
      - component.update: esptime

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# CRITICAL FIX: Optimized logger (major performance improvement)
logger:
  baud_rate: 115200  # Reduced from 2000000
  level: INFO        # Changed from DEBUG
  logs:
    # Reduce specific component logging
    truma_inetbox: INFO
    uart: WARN
    sensor: WARN
    binary_sensor: WARN
    web_server: WARN
    sntp: INFO  # Keep SNTP logs visible
    time: INFO  # Keep time sync logs

# CRITICAL FIX: WiFi optimizations
#wifi:
 # ssid: !secret wifi_ssid
 # password: !secret wifi_password
 # manual_ip:
 #   static_ip: !secret truma_ipaddress
 #   gateway: !secret motorhome_gateway
 #   subnet: !secret motorhome_subnet_mask
 #   dns1: !secret motorhome_dns1
 #   dns2: !secret motorhome_dns2
  # Performance optimizations
 # fast_connect: true
 # power_save_mode: none
 # reboot_timeout: 5min
 # enable_on_boot: true

ethernet:
  type: KSZ8081RNA
  mdc_pin: 23
  mdio_pin: 18
  clk:
    pin: GPIO16
    mode: CLK_OUT
  phy_addr: 0  # Usually 0 or 1, try 0 first
  manual_ip:
    static_ip: !secret truma_ipaddress
    gateway: !secret motorhome_gateway
    subnet: !secret motorhome_subnet_mask
    dns1: !secret motorhome_dns1
    dns2: !secret motorhome_dns2

api:
  encryption:
    key: !secret truma_api_encryption_key
  reboot_timeout: 15min
[... 345 lines cut ...]

  - platform: template
    name: "Set Electric Power Level to 1800 Watt"
    id: button_set_power_level_1800
    on_press:
      - logger.log: "Setting electric power level to 1800W"
      - truma_inetbox.heater.set_electric_power_level: 1800
      - delay: 500ms

# Perfect select config - keeping unchanged
select:
  - platform: truma_inetbox
    name: "Fan Mode"
    type: HEATER_FAN_MODE_COMBI
    id: select_fan_mode_combi

# CRITICAL FIX: Optimized web server (major performance improvement)
web_server:
  port: 80
  local: true
  version: 2
  include_internal: false  # CRITICAL: Major performance improvement
  auth:
    username: !secret web_username
    password: !secret web_password

santiniuk, 2025-11-05:
Oooohhh thanks so much, that looks a fantastic file. I'm also planning on a motorhome modification to my Truma system.

It looks like the code has pasted in a strange format. Is it possible to check please. I hope you don't mind, I plan on copying and amending but it looks like you have added some great comments. Time probs etc.

Thanks

zorrobyte, 2026-01-12:
Hello!

I added Aventa Aircon support and integrated other's fixes

https://github.com/Fabian-Schmidt/esphome-truma_inetbox/pull/67